MiSTer-devel/SNES_MiSTer #306: Sprite Time fetch test

paulb-nl, 2021-12-02:
This is a sprite time fetch test ROM for issue #303. There are 5 64x64 sprites on every row and 2 of the sprites have X position 256 so they are off-screen.

Row1:
2 sprites fully on screen + 2 off-screen = 32 tiles fetched so the last sprite only has 2 tiles visible.

Row2:
1 sprite is partially off-screen on the right (X position 232) and the SNES only fetches the visible 3 tiles. 2 sprites off-screen + 1 fully visible is 24 tiles. 3 + 24 = 27 so only 7 tiles left for the last sprite.

Row3:
1 sprite is partially off-screen on the left (X position -32) and the SNES only fetches the visible 4 tiles. 2 sprites off-screen + 1 fully visible is 24 tiles. 4 + 24 = 28 so only 6 tiles left for the last sprite.

Expected result:
[image]

SNES core release 20211126:
[image]

Test rom:
[sprite_time.smc.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/7642559/sprite_time.smc.zip)

Kopert, 2021-12-14:
I meant to test and report this earlier, but I just ran the test rom with `SNES_unstable_20211210_c8c9.rbf` and the results now match with the "Expected result" image.

paulb-nl, 2021-12-14:
Fixed with release 1214

closed 2021-12-14 by paulb-nl