MiSTer-devel/SNES_MiSTer #310: EMUDetect test rom showing picture of Dog in all initial wram settings.

SegaSnatcher, 2022-01-01:
Tested on Release 20211214.   I remember it use to show original hardware screen in a previous build.

[image]

birdybro, 2022-01-06:
https://github.com/MiSTer-devel/SNES_MiSTer/issues/184

Did you already try different initial WRAM states? Did you try to reset the system after changing different WRAM states?

Blitzwing1, 2022-01-08:
https://github.com/MiSTer-devel/SNES_MiSTer/blob/dfa6d823b30222dec356cf93e21d8848714630d8/rtl/CPU.vhd#L693

I thought maybe that this line in the recent commit may have been a case of a simple typo, changing it made the "EMUDetect" rom show as real hardware like it was doing before, but then the div_timing test rom which is passing in the two recent builds fails again.

birdybro, 2022-01-09:
Does anyone have the source to this test rom?

https://gitlab.com/higan/snes-test-roms

I'm still curious if it potentially fails on some real hardware revisions, as was indicated as a possibility the last time it was brought up.

Blitzwing1, 2022-01-09:
> Does anyone have the source to this test rom? I'm not seeing it in this extensive test rom suite that the higan/bsnes creator and crew have curated.
> 
> https://gitlab.com/higan/snes-test-roms
> 
> I'm still curious if it potentially fails on some real hardware revisions, as was indicated as a possibility the last time it was brought up.

Don't know. Just reading through #184 seeing as the EMUdetect test rom seems to only check for initial wram patterns on startup then it can't be very relevant to overall accuracy in games compared to the various tests the bsnes people put together.... By that I mean it isn't detecting for CPU, PPU or any other accuracy but just that the wram is right from a cold boot?

birdybro, 2022-01-10:
Google searching for it provides basically nothing other than one mention on smwcentral and the download link, it's really strange. Also, fat-shaming in ASCII, tsk tsk

[image]

:P

SegaSnatcher, 2022-01-12:
Just tested a build with latest commit.    Real Hardware detected now.   Thanks Srg320.   I will close when official build is released. 

[image]

closed 2022-02-02 by SegaSnatcher