MiSTer-devel/SNES_MiSTer #315: Audiable short beep before silent gameplay parts over Analog IO audio jack

Thelypody, 2022-02-28:
In the current release of SNES core (20220212) there's a very short audible beep being produced whenever the core stops playing sounds. Interestingly this beep is only present from Analog IO board's audio jack and not from DE10-Nano's HDMI port. Due to this I'm not discarding a possibility of hardware failure on my end. If no one else can reproduce this issue, is can be safely closed.
Going backwards though releases, it seems that all 2021 releases have this issue, but the sound seems to be much quieter. On the latest release the sound is loud enough that it made me finally notice it.
The fastest way to test it is to launch Super Mario World. A beep is audible right after the initial coin jingle.

Thelypody, 2022-03-29:
I'm glad to say that the release 20220326 is significantly less noisy than 20220212, taking it back to levels observed in 20220202 and before. The strange sound is still present but it is much more tolerable for ears.

sorgelig, 2022-03-31:
if you care about audio i suggest to use either optical output or HDMI. Analog output is basically poor-man's solution.

birdybro, 2022-05-22:
This issue can probably be closed, it was fixed a long time ago in the template core.

closed 2022-05-23 by sorgelig