MiSTer-devel/SNES_MiSTer #316: Analog audio noise

SWCDX2, 2022-03-02:
Hi,

There is noise coming out of the I/O board audio line out when specifically using the SNES core.

It is the same as M-Walrus described on this page: https://github.com/MiSTer-devel/SNES_MiSTer/issues/51

"...you can hear both faint static/noise at time and some times you can faintly hear some weird frequency modulation sounding noise (best i can describe it)."

To replicate the noise I'm hearing:

Load the game "Wario's Woods" > go to "VS Com" > pause the game

I've recorded the noise with my phone and uploaded it here:

SNES_MiSTer analog audio output noise 1

https://ln5.sync.com/dl/f9a270210/rgwmtbkv-yj3bimbb-umdqvay6-jt8z38a5

SNES_MiSTer analog audio output noise 2

https://ln5.sync.com/dl/2fbf51680/n22cykgj-5853vvte-zq695j4p-7qt8wrzp

birdybro, 2022-05-22:
This is expected. The analog IO board has more interference and the analog IO signal chain has almost no shielding at all. This is not a problem with the SNES core. This can probably be closed.

SWCDX2, 2022-05-23:
But of the few cores I've been playing with (GAMEBOY, Genesis, NES, SNES, TGFX16), it is the only one with this audio problem.

closed 2023-04-06 by SWCDX2