MiSTer-devel/SNES_MiSTer #319: Save issue: Zelda link to the past

bix66, 2022-03-15:
I can't get Zelda link to the past game saves to work on core SNES! I have tried them all but, after having performed the saves from within the game, if I try immediately I find the save I have all the elements I have found, the gems, the bombs, the arrows, etc. I turn off Mister FPGA, turn it back on, start the Zelda rom and find a previous save (by number of elements I found, gems, bombs, arrows, etc.), checking in the appropriate folder, the save shows data and time of the last save I made !! How is this possible ??? It seems that the new save can't overwrite the old one except as date and time .... Please help me to understand if it's my microSD problem or who knows what else or if it's a Core SNES problem. Help me by making a test you too with a Zelda link to the past rom on SNES and execute, randomly, some saves (I execute them with the select button of the joypad) and check when re-entering the game (both normally and after turning off and on the Mister FPGA) and kindly report what happens to you.
A thousand thanks

thehughhefner, 2022-03-16:
Do you have Autosave = On in the OSD? Because once you save in-game, you have to launch the OSD to save to mister sd card. I just tested this and no issues from my end.

bix66, 2022-03-16:
No, he hasn't activated Autosave = On. Zelda link to the past saves, the
game allows them at any time. But I try to insert the Autosave = On option
and see if it works. Thank you

bikerspade, 2024-09-25:
@bix66 I think this issue can be closed.

closed 2024-09-26 by sorgelig