MiSTer-devel/SNES_MiSTer #320: Star Fox Rom hacks

terminator2k2, 2022-03-30:
Some Star Fox Hacks just go to black screen and have been tested and working on SD2Snes + Super NT combo

https://www.romhacking.net/hacks/6402/

https://www.romhacking.net/hacks/6443/

and a soon to be released Star Fox EX Hack

kandowontu, 2022-03-30:
As the developer of said hacks, I am here if you have any questions or need an updated patch.

terminator2k2, 2022-03-31:
@kandowontu Star Fox EX worked on mister up to :

Star_Fox_Exploration_Showcase_8.85.SFC

anything after just black screen

birdybro, 2022-04-07:
It would be nice if a separate patch with the HUD could be made without the speed hacks. The MiSTer core has Super FX turbo and CPU Turbo which do speed up the game considerably already, so if you want to play the game with a smoother framerate for now, that might help.

jeffkrull, 2022-10-12:
If it's any help, it seems that the problem is only with the 21 MHz FastROM hacks. The 21 MHz hack without FastROM works just fine in the SNES core.

protogem2, 2023-03-26:
SuperFX FastROM hacks don't seem to work at all, as Stunt Race FX's FastROM hack also doesn't boot.

kandowontu, 2023-04-30:
> SuperFX FastROM hacks don't seem to work at all, as Stunt Race FX's FastROM hack also doesn't boot.

Stunt race FX fastrom DOES boot and work. You must have a bad patch (probably from the sfc speed hacks ia page).

closed 2024-10-20 by terminator2k2