MiSTer-devel/SNES_MiSTer #321: Feature Request: usb2snes support

mrchrislad, 2022-04-05:
Would adding usb2snes support to the SNES core be possible?

This is the main thing keeping my SNES and FXPak Pro wired up, as I use it for autotracking and multiworld syncing with the ALttP and Super Metroid (and both together) randomizers.
I'm sure the code that the FX Pak/sd2snes uses is here, https://github.com/RedGuyyyy/sd2snes usb branch, but I wouldn't even know where to begin implementing it.
Is there someone much cleverer than me that would be able to have allok if it is possible?

evidencedj, 2022-04-05:
There is already a feature request for usb2snes in the Github Issues :)

https://github.com/MiSTer-devel/SNES_MiSTer/issues/276

mrchrislad, 2022-04-05:
Ah ha, I knew I wasn't alone! I searched for usb2snes in issues and nothing flagged up. Hoping this gets impletmented at some point!

closed 2022-04-05 by mrchrislad