MiSTer-devel/SNES_MiSTer #324: Bad interlaced resolution.

PepAlacant, 2022-04-10:
Rendering Ranger Still doesn't work, and there is a resolution issue in interlaced modes. It can be noticed in "Super Family Tennis". The selection mode is in high resolution interlaced, and it's shown incorrectly in the CRT resolutions.

birdybro, 2022-04-10:
Yes, there is still an issue open for Rendering Ranger here --> https://github.com/MiSTer-devel/SNES_MiSTer/issues/292 - It's not closed yet.

For the interlaced resolution of Super Family Tennis... It stays on 512x224 with an occasional rare switch to 512x448i on screen transitions before the title screen. When using a RetroTink5x Pro to get the lines and frequency, it looks correct to me. 525i input lines (this is correct for NTSC), 15.75kHz input h-freq, and 59.98hz v-freq (correct for SNES).

**RetroTink 5x Pro capture from MiSTer**
[image]

**HDMI Capture from MiSTer**
[image]

And in game shots are the typical 262p as usual (480i is actually 262.5p --> 525i for reference).

PepAlacant, 2022-04-10:
I'm using a Sanyo TV with SCART. It is a PAL TV but it has support for NTSC modes. The game shows ok in the Super Famicom and in the MiST FPGA SNES core. But it shows really strange in the MiSTer core.

PepAlacant, 2022-04-10:
I'll do the test with a PVM tomorrow and I'll try to get an image.

PepAlacant, 2022-04-10:
[image] it's shown interlaced, but it shows as it. Wrong resolution.

PepAlacant, 2022-04-10:
My fault ! I had a the 256px option enabled ! Sorry so much.

closed 2022-04-10 by PepAlacant

birdybro, 2022-04-10:
Yup! I just was about to say that was probably what was happening. :) Glad you figured it out!

PepAlacant, 2022-04-10:
sorry for the inconvenience !