MiSTer-devel/SNES_MiSTer #325: SNAC issue with NTT Data controller in Super Mario World (USA)

M-Walrus, 2022-04-12:
The NTT Data controller does not respond in Super Mario World, though it does on original hardware.  

This controller is known to have compatibility issues in various games even on original hardware, like Super Mario Kart where it will repeatedly jump the entire time you are racing.  Super Mario World should work correctly with it though.  If possible, a "NTT Data compatibility mode" option would be a good addition to the core (Analogue's Super NT console has this), as it would allow all games to function properly with this controller.  Many people use this controller on MiSTer.

There may be issues with other games in the library when used with this controller, but this is the only one I know of so far where it doesn't match original hardware behavior.

birdybro, 2022-04-12:
That's interesting that it doesn't work.

https://www.raphnet.net/divers/ntt_data_sfc_controller/index_en.php

Generally speaking, the extra buttons are the only things that should be unable to work, there shouldn't be any differences in behavior. I think I'll order an SNES SNAC adapter and hook up signaltap to get some more information to provide here later, which hopefully could help someone else fix it.

birdybro, 2022-04-19:
So got a SNAC adapter, tested it out. blue212 is helping me learn signaltap to see what is going on. From his advice holding 0 on the NTT and then pressing buttons works in the title screen and overworld map, and then within stages it works normally. I wonder if holding 0 was required on original hardware? Maybe some chip variation of the SNES caused different behavior? Not sure. Going to look into it more later on, it's possible a status bit could be used to keep bit 13 high despite the NTT pad trying to keep it low, then it would signal as a regular gamepad and disable the extra keys and improve compatibility maybe.