MiSTer-devel/SNES_MiSTer #326: Feature request: Add Super Gameboy games

zal16, 2022-04-19:
It's possible to add Sharp LR35902 (Z80 clone) simulation to archieve compatibility to Super Gameboy games?

birdybro, 2022-04-19:
There isn't a whole lot of space left in this core. This is one of the largest cores in the entire project currently. There is already Super Game Boy functionality within the Game Boy Core (albeit lacking the SNES hardware functions that are used in a tiny handful of games).

birdybro, 2022-05-25:
https://misterfpga.org/viewtopic.php?f=20&t=4786 - currently unofficial test build of a core that does this with MSU-1 support added as well, if you want to try it out @zal16

wwark, 2022-07-26:
The core Super Game Boy is now released.
This ticket could be closed.

closed 2022-07-27 by sorgelig