MiSTer-devel/SNES_MiSTer #327: Sound issues with Bishōjo Senshi Sailor Moon S: Jōgai Rantō!? Shuyaku Sōdatsusen

mightysquirrel, 2022-05-02:
O-Range from the Sailor Moon S community, noticed that sounds 4 and 5 in the sound test, sound identical to the Wii Virtual Console emulator. The sounds do not sound the same as original hardware.

birdybro, 2022-05-02:
Do you have a recording off of original hardware to help diagnose the problem?

mightysquirrel, 2022-05-02:
I do! 

https://media.discordapp.net/attachments/497126993616437249/968168447269822474/sound_effects.mp4

birdybro, 2022-05-22:
I have a couple examples from MiSTer to compare for reference. This is with no filter:

https://user-images.githubusercontent.com/16388068/169717805-a5d432f1-a004-461a-9ee7-eedfa62fbe62.mp4

This is with the GPM-02 filter:

https://user-images.githubusercontent.com/16388068/169717815-346d00fd-b9b8-4d83-b9e6-b8b50eaf750e.mp4

Definitely sounds different, mostly on the second example. Sound effect 05.

MP2E, 2024-10-02:
Just compared the latest SNES core nightly to the original hardware recording and the sound effect sounds identical to real hardware now! Excellent work srg320!

paulb-nl, 2025-06-29:
Can be closed.

closed 2025-06-29 by sorgelig