MiSTer-devel/SNES_MiSTer #329: Feature Request: "Show Everything" Mode

mjj03301977, 2022-05-12:
Rysha has work-in-progress build of the NES core with a "show everything" mode. This will allow the core to draw everything to the screen in the border/overscan area untouched. This feature is very much wanted by myself (and I am sure a few others) because some games actually showed content in this area. Plus you get potentially get colored borders, screen flashes, garbage, etc. It is a nice option to have for those who do not want to cut out this area or have it draw a black border.

I am now hoping that some day we can get the same in the SNES core. I have read that possibly three (or more?) games actually use this border/overscan area:

Rendering Ranger R2
Tetris & Dr. Mario
Pilotwings (Europe): overscan occurs after you play a mission and seems to stay.

Here is a thread about it...
https://www.reddit.com/r/miniSNESmods/comments/75bkmq/how_to_disable_overscan_in_certain_games_to_make/#bottom-comments

Thank you for all your hard work on these amazing cores!!!

PS: Obviously if this happens, then aspect ratios will have to be re-calculated I am guessing...using Rysha's calculator.

paulb-nl, 2022-05-13:
Overscan enabled on the SNES results in the 256x239 resolution and this is already supported. 

The European Super Mario World also uses this mode.

Did you test the games you mentioned?

birdybro, 2022-05-13:
Chrono trigger also has overscan shown in some graphics modes where there is a big gap of black in the bottom, and this is already accurately portrayed in the core.

mjj03301977, 2022-05-13:
My apologies…I thought the core always showed black borders no matter what. Before I close this, I have one last question (not able to test since I’m at work).

Is there any way to show 256x240 or 512x240? I believe the SNES actually output 240 instead of 239.

mjj03301977, 2022-05-13:
Byuu (RIP) talks about it here

https://archive.nes.science/nesdev-forums/f2/t10141.xhtml#p112837

birdybro, 2022-05-13:
Ah yeah... I vaguely remember reading about this a long time ago ;)

https://wiki.superfamicom.org/registers#setini-screen-modevideo-select-640

It looks like if the game has bit 2 of register `SETINI ($2133)` set to `o` then it's in overscan mode, and 239 lines are displayed. bsnes showed 240 lines, so this might be worthwhile to have as an option to have parity with bsnes (and ostensibly original hardware).

paulb-nl, 2022-05-14:
Original hardware outputs 224 or 239 lines. Byuu included the pre-render line but there is no output on that line so an option for 240 lines makes no sense.

mjj03301977, 2022-05-14:
So I’m testing and there is no overscan toggle in this core. Only 512x239 or 256x239 if force 256 is enabled. Are we positive the black borders are being drawn to the screen? I don’t think they are…
Definitely not if I compare to something like the Genesis core.

mjj03301977, 2022-05-14:
[image]

Here is what Wikipedia shows (assuming it is correct). I will just close this for now. I think I was just misunderstanding and comparing the SNES core to the NES core too much.

closed 2022-05-14 by mjj03301977