MiSTer-devel/SNES_MiSTer #331: [MSU] Heartthrob Memorial: Under the Tree of Legends (Tokimeki Memorial Translation and MSU hack) - Konami voiceline missing & Intro is blacking out from time to time.

Hobbesgoblin, 2022-05-17:
Expected behaviour (from snes9X emulator):
After the Konami jingle a voiceline saying "Konami"
Intro is playing normally.

Seen behavious on MiSTer:
Voiceline is missing
Intro blacks out for a moment a twice in the beginning. (consistent at the same places before the title card)

sorgelig, 2022-05-17:
"Konami" voice is missing, yes, but i don't see any issues in intro video

Relikk, 2022-05-17:
I can confirm dropouts in the video on my set-up, also. From SD card. We were comparing with Kitrinx on the Discord with Road Avenger. Their intro video was perfect and they mentioned they were loading from their Network, but a few of us loading from SD card were getting an error.

sorgelig, 2022-05-17:
data track is preloaded fully at rom loading. So it doesn't matter network or SD card.
Can someone make a video clip showing the problem?

Hobbesgoblin, 2022-05-17:
Am also loading from SD card.
I dont have any capture equipment therefore phonecam. Poor quality, i know, but dropouts are visible

https://user-images.githubusercontent.com/47921486/168910778-d0dad977-6213-4c94-ba2c-a74c16e23189.mp4

.

birdybro, 2022-05-17:
Does this occur when you switch vsync_adjust to 0 in the MiSTer.ini?

Hobbesgoblin, 2022-05-17:
Only oaVideo was taken hat  vsync_adjust=0
Just tested all 3 modes.
it ovvurs at  vsync_adjust =0 and  vsync_adjust = 1
it does NOT occur at  vsync_adjust = 2

birdybro, 2022-05-17:
The ROM is likely producing video that is out of spec for the SNES. The MiSTer isn't a software emulator so you can't get away with hacks like that. At least this is my assumption of what is happening given that vsync_adjust=2 works, and the others don't.

sorgelig, 2022-05-17:
this should fix video problem:
[SNES.ZIP](https://github.com/MiSTer-devel/SNES_MiSTer/files/8712012/SNES.ZIP)

it doesn't fix absence of "konami" voice.

Hobbesgoblin, 2022-05-18:
> this should fix video problem: [SNES.ZIP](https://github.com/MiSTer-devel/SNES_MiSTer/files/8712012/SNES.ZIP)
> 
> it doesn't fix absence of "konami" voice.

Yes this fixes the dropout issues in the video on vsync_adjust = 0 and 1

hackgrid, 2022-11-09:
I just wanted to add, that while bsnes plays this "Konami"-sound fine, sd2snes/fxpak on a real hardware snes also misses this voiceline!

So maybe Mister is "correct" there? Or there is the same bug in sd2snes and Mister cores.

Relikk, 2022-11-09:
No, it's a patch issue. MSU1 implementation in MiSTer is still buggy, but this issue is not normal or intended behaviour. If the jingle plays on an emulator and not on an SD2SNES or MiSTer, or whatever, then there is a coding error or a missing command in the patch source.

Relikk, 2022-11-10:
So, yes. There is no wait loop on further inspection with a debugger.

7cee17 ldx #$0001          A:00ff X:0001 Y:96dd S:1fc0 D:0000 DB:81 N.M..I.C V:236 H:196 F:28
7cee1a stx $2004            [812004] A:00ff X:0001 Y:96dd S:1fc0 D:0000 DB:81 ..M..I.C V:236 H:202 F:28
7cee1d plx                       A:00ff X:0001 Y:96dd S:1fc0 D:0000 DB:81 ..M..I.C V:236 H:211 F:28
7cee1e lda #$01              A:00ff X:0001 Y:96dd S:1fc2 D:0000 DB:81 ..M..I.C V:236 H:220 F:28
7cee20 sta $2007            [812007] A:0001 X:0001 Y:96dd S:1fc2 D:0000 DB:81 ..M..I.C V:236 H:224 F:28

But any time a byte is written in an attempt to fix it, the checksum is bad and it doesn't boot anymore. CRC Fix should make it work, but it still doesn't boot.

Best to report it to the authors and let them sort it out.

hackgrid, 2022-11-10:
Thanks for checking and trying to create a fix! :)
Unfortunately I dont know how to contact them, they seem pretty closed off?
Hopefully somebody knows how to contact :)

paulb-nl, 2025-06-29:
Issue can be closed. Bad hack issue.

Zazerek, 2026-10-21:
> 12\. So, yes. There is no wait loop on further inspection with a debugger.
>          7cee17 ldx #$0001          A:00ff X:0001 Y:96dd S:1fc0 D:0000 DB:81 N.M..I.C V:236 H:196 F:28
>          7cee1a stx $2004            [812004] A:00ff X:0001 Y:96dd S:1fc0 D:0000 DB:81 ..M..I.C V:236 H:202 F:28
>          7cee1d plx                       A:00ff X:0001 Y:96dd S:1fc0 D:0000 DB:81 ..M..I.C V:236 H:211 F:28
>          7cee1e lda #$01              A:00ff X:0001 Y:96dd S:1fc2 D:0000 DB:81 ..M..I.C V:236 H:220 F:28
>          7cee20 sta $2007            [812007] A:0001 X:0001 Y:96dd S:1fc2 D:0000 DB:81 ..M..I.C V:236 H:224 F:28
>          But any time a byte is written in an attempt to fix it, the checksum is bad and it doesn't boot anymore. CRC Fix should make it work, but it doesn't boot.
>          EDIT: The checksum can be fixed but it still doesn't boot. Speculation that there was a tamper check in the code somewhere, but nothing has been found.
>          Best to report it to the authors and let them sort it out.

I was looking into this issue today and found the checksum routine at 0x7DE8 to 0x7E0D in the ROM with the help of a debugger.
I changed 0x7DF8–0x7DF9 and 0x7E09–0x7E0A to the new checksum values and it booted in both emulator and real hardware via FXPakPro.

I wrote a new routine at 0x7E6E80 with 2C 00 20 70 FB A9 01 8D 07 20 6B
BIT $2000
BVS $7CEE80
LDA #$01
STA $2007
RTL

And replaced all instances of A9 01 8D 07 20 with 22 80 EE 7C EA (to jump to my new routine)
JSL $7CEE80
NOP

And changed the checksum routine
0x7DF8–0x7DF9 set to 80 63
0x7E09–0x7E0A set to 67 C2

I didn't bother fixing the CRC cause it still boots on real hardware anyways
I hope this helps someone googling a fix cause thats where I was at this morning
I don't own a Mister but this did work for me on real hardware, tho I only tested Yuinas ending because thats all I have unlocked

hackgrid, 2026-10-21:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Awesome, great work! :)

Could you supply a bps patch for this?

Zazerek, 2026-10-21:
@hackgrid 
Try this. Patch over the original English patched ROM
[https://drive.google.com/file/d/1IYM1N97_bm5rYnQlbp-2ON8vyCyHzUKS/view?usp=sharing](url)