MiSTer-devel/SNES_MiSTer #335: Stunt Race FX runs slower than real hardware

AtomicShroom, 2022-05-27:
The 3D portions of Stunt Race FX handled by the FX chip are running significantly slower than real hardware as reported by [Storminormin](https://misterfpga.org/memberlist.php?mode=viewprofile&u=248) on the forums:

https://youtu.be/FrbCKzAopqE

Left is real game on NA-NTSC SNES (2chip GPM-02), right is MiSTer.

2D elements handled by the SNES like the timer run at the same speed however.  Might be worth investigating other FX games as well. Something is definitely off.

AtomicShroom, 2022-05-27:
Video from another user ([Storminormin](https://misterfpga.org/memberlist.php?mode=viewprofile&u=248)) comparing an actual Wild Trax (JP version) cartridge to MiSTer:

https://www.youtube.com/watch?v=TK0DhknO6co

Same conclusion:  The 3D logic/render is running slower than real hardware.

AtomicShroom, 2022-06-06:
Not sure if related, but the Yoshi's Island world map screen also appears to be running way slower than real hardware. I figure it might be related since it uses the Super FX2 chip:

https://youtu.be/5Nd4iAfiNK0

I don't have a real cart to compared with anymore, so instead I used footage from an AGDQ speedrun, which (I would hope) don't let runners play on anything other than real hardware.  The AGDQ footage seems correct based on my memory from playing this game about a bazillion times.

sdufus, 2022-06-07:
Here's a comparison I made between Yoshi's Island on MiSTer using the latest public SNES core and an original 1chip ntsc console with actual cartridge.

https://www.youtube.com/watch?v=GXI-ftEh13k

The MiSTer implementation is slower here as well.

paulb-nl, 2022-06-17:
I made a test #340 to verify the Super FX speed which should help to improve the core.

AtomicShroom, 2022-09-07:
Based on my memory it looks fixed in the latest unstable build! 🥳

@sdufus could you recompare to make sure?

sdufus, 2022-09-07:
Actually just did a new comparison of Wild Trax with the SNES_unstable_20220903_16d615 nightly less than 13 hours ago
https://www.youtube.com/watch?v=fQK2OOFPF-k

From the test I did (and a few other recent ones) it's definitely improved. Seems to be off about 30 milliseconds according to the in-game clock. But that's after about a minute of playing. Don't know exactly how much it'd be off after extended play.

Suppose it could also be an issue with the age of components on my console. It was suggested in a youtube comment I try a comparison with a non-1-chip snes to see if there's a difference, and I'd like to, but don't have one to test with.

AtomicShroom, 2022-09-09:
Alright well I’ll close this issue for now. I think we should open a new issue should we conclude there’s still a difference down the line. During gameplay it’s indistinguishable.

closed 2022-09-09 by AtomicShroom