MiSTer-devel/SNES_MiSTer #336: MSU: Contra III - The Alien Wars: Music tracks appear to be off by one

c0d3h4x0r, 2022-05-30:
When the Konami logo appears at power-on, the level 1 background music plays.  When you start playing level 1, the level 1 boss fight music plays.  While fighting the level 1 boss, the short victory music plays over and over again.  It seems the music tracks are off-by-1.

c0d3h4x0r, 2022-05-30:
In fact, this off-by-one thing appears to be a global issue with all MSU-enhanced games I've tried.

Kitrinx, 2022-05-30:
I can't reproduce this issue in contra III or any other game. Possibly you have a bad set of roms?

c0d3h4x0r, 2022-05-30:
Over 50 ROMs all misconstructed in the exact same way?  That's beyond unlikely.

Kitrinx, 2022-05-30:
It is unlikely, but it's more likely than nobody else of many thousands of people mentioning this egregious issue. Check the indexing of the audio files and perhaps compare them to another set. Make sure you're using the latest unmodified main and SNES core from the mister repo. Perhaps have someone else test one of your msu1 roms to see if it's the data.

sorgelig, 2022-05-30:
check if there is a specific map file which redefines index to file name. So romset may relay on mapping which MiSTer doesn't use.

Kitrinx, 2022-05-31:
You can review the correct naming conventions here:
https://www.zeldix.net/t1607-msu1-getting-started-guide

c0d3h4x0r, 2022-05-31:
There are several different naming conventions listed there.  Which convention is correct for MiSTer?

c0d3h4x0r, 2022-06-01:
Confirmed that a nee fileset renamed per the sd2snes convention works as expected.  So this issue is really just a matter of insufficient/incomplete wiki documentation.

bikerspade, 2024-09-25:
> Confirmed that a new fileset renamed per the sd2snes convention works as expected. So this issue is really just a matter of insufficient/incomplete wiki documentation.

@c0d3h4x0r can you close the issue?