MiSTer-devel/SNES_MiSTer #338: "SNES/Hardware/CPU Speed: Turbo" setting is not persistent when LOAD a new ROM

javier-rodas, 2022-06-09:
1) LOAD SNES core
2) OSD => SNES/Hardware/CPU Speed => Turbo
3) OSD => System => Save settings
4) RESET the MiSTer
5) LOAD SNES core
6) OSD => SNES/Hardware/CPU Speed => We verify that the setting is still in Turbo mode => Correct!.
7) OSD => Load *.SFC => Select any game
8) OSD => SNES/Hardware/CPU Speed => This option is set automatically back to NORMAL => Incorrect!

The "CPU Speed: Turbo" setting is persistent when using "Save settings" among system reset, but is set to NORMAL by default when load any ROM, so the "Save settings" is not actually taking effect on this specific option.

sorgelig, 2022-06-11:
It was made on purpose

javier-rodas, 2022-06-12:
Is there any way to bypass this and make this parameter persistent among LOAD of ROMS too?.

Maybe you could explain us which lines of code we could patch to have a version of the core which is coherent with the settings that the user manually save (using the "OSD => System => Save settings" option). I mean, to save all the settings, including this "CPU Speed: Turbo" too. Thanks.

sorgelig, 2022-06-12:
remove or comment-out this line:
https://github.com/MiSTer-devel/SNES_MiSTer/blob/a841f8132ec15b0477d4d898daf776da33d17c48/SNES.sv#L403

Many games misbehave when turbo is activated. So user may set turbo and forget. Then after some time start to complain about non-working games.

javier-rodas, 2022-06-13:
Thanks for your reply. I fully understand your concern when Turbo is not really well supported by all ROMs and some users may complain. But, on the other side, there are some few games, for some users like me, that are nearly "unplayable" without the "CPU Mode: Turbo". If we always want to play maybe 1 to 3 favourite SNES games, and if all of them work well with Turbo, it's just convenient to have an SNES core version which respects that "save settings" option when the ROM is loaded, right?. Those ROMS could be manually loaded, automatically loaded with @pocomane MBC: https://github.com/pocomane/MiSTer_Batch_Control or maybe automatically loaded during boot (after powering the MiSTer, without any need to go into the OSD to enable that Turbo mode each single time). These are the scenarios that I mean it may make sense to respect those "save settings" including the "CPU Mode: Turbo".

By the way, can I ask you it it's already possible or if you are maybe planning to provide any mechanism to give "parameters" when launching a MiSTer core? That way it would be maybe possible to launch the SNES core specific settings (such as the Turbo enabled or disabled by default) without the need to recompile/fork the core, which always creates undesirable fragmentation on the MiSTer project itself. Thanks.

sorgelig, 2022-06-13:
I have some thoughts about this. It's not as easy as it looks at first sight. It also adds restrictions how core may change existing options later which is undesirable side effect. May be later it will be implemented

javier-rodas, 2022-06-13:
Ok, thanks.

closed 2022-06-13 by javier-rodas