MiSTer-devel/SNES_MiSTer #339: Feature Request: Volume Boost for MSU1 Games

thehughhefner, 2022-06-14:
Hi,
There are msu1 games that suffer from having low volume. This is apparently a known issue (even for SD2SNES) and how SD2SNES remedies this is by having a "Volume Boost" option that increases the volume of the MSU1 track. It does increase it in steps (similar to what we have in the pc engine core).

birdybro, 2022-06-17:
[SNESmsu1boost.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/8924655/SNESmsu1boost.zip)

Test this out for me, the option is in Audio & Video. Keep the volume low when testing at first, some games might be too loud, I didn't test a lot of games.

birdybro, 2022-06-17:
On second thought, this feature shouldn't be added (after talking to Kitrinx and sentient6 about it in detail).

https://www.zeldix.net/t1195-correct-msu-1-volume-levels

Apparently a lot of MSU-1 hacks need to be adjusted still, bsnes had a sort of boost for it (bsnes was the source of MSU-1).

thehughhefner, 2022-06-18:
Hi,
I'm not really sure if I understand why this feature shouldn't be added. I wonder what you discussed about. 
It's true as you say that lots of MSU1 hacks needs to be adjusted in terms of volume. And as you say "There's no way to tell what is "true" without someone showing us the assembly code in a hack and demonstrating that it is not in accordance with bsnes' spec.". This is the primary reason why it is good to have a volume boost option in the core as a quick fix (for lack of a better term). As a user of a SD2SNES, the volume boost option does a really good job 

I would love to have an opinion from developers or people that has more insight into MSU1. We have it in SD2SNES/FX pak pro as well as bsnes.
Kind regards

sorgelig, 2022-06-19:
try this build:
[SNES.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/8935561/SNES.zip)
in "Audio & Video" i've added MSU boost option.

thehughhefner, 2022-06-19:
Hi,
Thank you so much @sorgelig. This is exactly the feature release I asked for. 
I hope this can get merged!
Kind regards

birdybro, 2022-06-19:
@thehughhefner read the thread I linked. If sorg wants to add it that's fine. I started to add it, but decided I personally wouldn't look into it anymore because the hacks that don't have proper audio balance should be fixed by the people who made the hacks. The reason why sd2snes had it was because sd2snes for 4 revisions had varying audio problems. The reason why bsnes didn't have it is because that's how it was intended. The reason why there are varying audio balance in hacks is because people were trying to work around the design flaw in sd2snes, the target for MSU1 hacks shifted from bsnes to sd2snes, which had a hardware design issue for years.

SegaSnatcher, 2022-06-20:
@birdybro 

A lot of those hacks have been around for years and honestly I don't expect any of the original creators to fix them, which is why having this volume control would be very helpful.  You never know what you will find when searching MSU-1 packs online.  It's a mess unless you have the skills to build these MSU-1 hacks yourself, which most probably don't.

thehughhefner, 2022-06-20:
> @birdybro
> 
> A lot of those hacks have been around for years and honestly I don't expect any of the original creators to fix them, which is why having this volume control would be very helpful. You never know what you will find when searching MSU-1 packs online. It's a mess unless you have the skills to build these MSU-1 hacks yourself, which most probably don't.
> 
> In a perfect world we'd have properly adjusted volume for all MSU-1 hacks, but realistically its not what most people will find when searching for MSU-1 packs.

Exactly!

Toryalai1, 2022-06-20:
> try this build: [SNES.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/8935561/SNES.zip) in "Audio & Video" i've added MSU boost option.

I have tested this with multiple msu1 hacks and it really does wonders! Thank you :)

closed 2022-06-20 by sorgelig

Relikk, 2022-06-21:
> A lot of those hacks have been around for years and honestly I don't expect any of the original creators to fix them

Just chiming in. This option is probably there for good now, but there's no need to search for "packs" when the vast majority of MSU1 hacks originate at Zeldix, and that's where they are still maintained and have been most likely fixed since. These packs can contain outdated patch versions and unfixed audio, hence the complaints, not to mention the audio options are usually cherrypicked. By the time you've waited on Archive to finish your download, you could have downloaded the patch and audio from Zeldix, sourced the base ROM and patched it to the latest version in the meantime.

I'm not entirely against the idea of having a volume boost option in the core for MSU1, as people will prefer audio to be either louder or lower depending on their tastes (I know I personally strive to create audio for it to not be overbearing or drown out the SFX and to be as close as I can to the original SPC volume), but if people simply updated their patches or audio packs from source then I think the boost option wouldn't be necessary.

sorgelig, 2022-06-21:
I will revise volumes in MSU1 mode.

SegaSnatcher, 2022-06-21:
> I will revise volumes in MSU1 mode.

Will there still be an option to boost audio for those that want louder music?   Now that it was officially added, I hope the boost stays for good.  Thanks.

thehughhefner, 2022-06-22:
Thank you sorgelig for revising the volumes for MSU1. I agree with @SegaSnatcher, I hope that the boost option stays in for good. It is a togglable option anyway.