MiSTer-devel/SNES_MiSTer #341: MSU1: PCM looping issues

Relikk, 2022-06-17:
There is a minor looping issue with PCM files. The exact loop point specified in the files is not being triggered correctly, and usually sounds like it's earlier than expected, resulting in an audible pop due to a lack of reasonable zero crossover at the loop point. Using Turtles In Time as an example as the character select track is fairly short, here are some waveform comparisons from my original source file and from the MiSTer output.

**Source:**
[image]

**MiSTer:**
[image]

The WAV files from the screenshots:
[TMNT.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/8927700/TMNT.zip)

The start of the loop looks like it is being cut off.

There is also still some track loop issues in the core where tracks that are specified to loop in code, do not loop. Using the mode select screen on Fatal Fury Special as an example _(track 18 dec., $12 hex)_, in the ASM for the MSU1 patch it is specified to loop.
[image]

**Source files used for both examples:**
TMNT patch and arcade PCM files: https://www.zeldix.net/t1443-tmnt-iv-turtles-in-time
Fatal Fury Special patch and Neo Geo/Arcade PCM files: https://www.zeldix.net/t2193-fatal-fury-special

protogem2, 2023-11-26:
Could this issue perhaps be addressed/fixed? It is causing quite a few issues with looping on MSU-1 pcms.

paulb-nl, 2024-09-23:
Thanks for the examples and links to PCM files. Should be fixed with commit 4955bdc8d02c74c1c5ed483a9a1af485b1efbbaa