MiSTer-devel/SNES_MiSTer #343: Link to the Past SA-1 Practice ROM does not load

40wattrange, 2022-06-27:
Not sure if this is something that would be addressed but there is a practice ROM for a Link to the Past that has been modified to make use of the SA1 chip. It seems that it will not load. I have attempted using the various memory settings (SNES, SNES2, SD2SNES, FF) but all results are the same.

You can find the ROM here: https://spannerisms.github.io/lttphack/

Again, not sure if this is something that would even be considered, but it will work on a real SNES using SD2SNES, so figured I'd post it in case it was something that could/should be addressed.

birdybro, 2022-06-27:
256kb or 256KB? Huge difference between the two. 256KB is the max supported limit of BW-RAM in the SA-1.

willk, 2023-10-29:
> 256kb or 256KB?
The practice hack's site says "kb": [https://spannerisms.github.io/lttphack/](https://spannerisms.github.io/lttphack/). They discuss more details SA-1 and implementation about 1/8th down the page.

littlegamer87, 2024-09-21:
srg recently updated core with wram reallocated (amazing!)
Gave this rom a test, runs fine it seems

bikerspade, 2024-09-25:
@40wattrange can you re-test with the updated core, and close this issue if it is now resolved?

paulb-nl, 2025-01-22:
This is fixed with commit 74d7761d549f5bb81ef15d30bf5e136edbc0ebaa

closed 2025-01-23 by sorgelig