MiSTer-devel/SNES_MiSTer #344: Graphics glitch Secret of Mana with Scandoubler HQ2x enabled

littlegamer87, 2022-07-04:
See screenshot below. Pretty sure this is a regression and used to work fine, now running MiSTer v220529 with SNES v220624, USA ROM w/CRC d0176b24

Glitch appears to be related to scandoubler HQ2x. Disabling it removes it.

[image]

birdybro, 2022-07-04:
Do you have force 256px turned on in the OSD? If not, does enabling it combined with the scandoubler get rid of it?

sorgelig, 2022-07-04:
there is some regression. I will investigate it

littlegamer87, 2022-07-05:
> Do you have force 256px turned on in the OSD? If not, does enabling it combined with the scandoubler get rid of it?

Force 256px was disabled. Enabling it does result in the full screen being rendered, but the effect on the text is not pleasant in my opinion. See screenshot attached, letters like M, N, W all look like an H.
[image]
t on menu text is not pleasant in my opinion.

birdybro, 2022-07-05:
Yup, it's expected for the text to get garbled, was just curious. Thanks. :)

littlegamer87, 2022-07-14:
Confirmed as resolved in release [20220712.](https://github.com/MiSTer-devel/SNES_MiSTer/commit/198440a423264178f6ac6c630d260936ade50171)

Thank you for the amazing support!

closed 2022-07-14 by littlegamer87