MiSTer-devel/SNES_MiSTer #345: Intro Music Slowdown in Soulblazer

leonquest, 2022-07-12:
Intro Music, or seemingly the whole gane, slows down in the game intro, at the beginning when the letters scroll and the intro music crescendo starts.

I tested the same Soulblazer no-intro US rom on real hardware, an SHVC revision and a 1chip, and the intro plays normal without slowdown.

As a side note, this is a common bug on most emulators I've used over the years, try it on zsnes and others and you will find the same bug. I have not tested on the latest version of Higan.

srg320, 2022-08-16:
Use menu `Hardware`->`Initial ARAM`. The best option for this game is `9966(SNES2)`.

birdybro, 2022-08-16:
@leonquest here is a download link tot he latest unstable build with this change --> https://github.com/MiSTer-unstable-nightlies/SNES_MiSTer/releases/download/unstable-builds/SNES_unstable_20220816_14f8ad.rbf

mrnarse, 2024-04-14:
Just booted this game up on the MiSTer yesterday and was surprised that the audio for this game was playing _without_ slowdown, which led me to find this issue.

When I played this game on original hardware (SHVC-CPU-01) with an original US cartridge back when the game was released, the audio slowdown was definitely present at the beginning of the intro.

I'm sure this is accurate to later hardware revisions of the SNES, but I'm wondering if there's any way to configure the MiSTer core to replicate the slowdown and be accurate to the original SHVC-CPU-01 behavior again?

leonquest, 2024-04-14:
I have an SHVC-CPU-01 and there is no slowdown on the real console, as I stated 2 years ago

closed 2024-04-14 by leonquest

mrnarse, 2024-04-14:
Guess the SHVC-CPU-01 wasn't the most reliable hardware.
[image]
Some further info about the issue from when it was brought up after the Super NT's release 6 years ago, should this issue ever be re-opened.