MiSTer-devel/SNES_MiSTer #348: Phantom 2040 - random inputs during lag frames

developwisely, 2022-08-19:
I'm a speedrunner of Phantom 2040 and there's some issues with the inputs that I have come across.  When mashing during heavy lag frames, random inputs will start firing.  This does not happen on console.

One example: https://drive.google.com/file/d/1fJ9wnh_I9YBgdWkxwxiFaQ-BKQ8Ao921/view?usp=sharing

In this example, when trying to use spread to lag out the game, you can see from the input viewer that I'm only mashing Y and X, however you can also see that the rope comes out (green line), which is an A input.

There are other times where START will spam, which pauses the game and brings up the menu, instead of A even though it was never pressed.  If you need more examples I can try and get some video of them.

birdybro, 2022-08-19:
Are you using a USB controller, BliSSter, or SNAC?

birdybro, 2022-08-20:
Oh just realized you are probably using a RetroSpy ;)

I wonder if the core is slightly more sensitive than a real system. Does the same thing occur without the RetroSpy hardware involved?

developwisely, 2022-09-19:
I'm using Super Famicom controller + Raphnet adapter.  Yes, confirmed that it still happens without RetroSpy or any sort of input capture software on.

The inputs that it decides to fire seem random, sometimes it will be Start, sometimes A, sometimes B, etc.

birdybro, 2022-09-20:
Can you test what the controller does with the two snes test roms for me?

https://tcrf.net/SNES_Test_Program

https://tcrf.net/SNES_Burn-In_Test_Cart

It almost sounds like a multiplexing issue or something...

developwisely, 2022-09-20:
Here's the controller tests from both roms: https://drive.google.com/file/d/12cwF2gkB9VLA7N9hQSHeSxTcP6pC-bXg/view?usp=sharing

paulb-nl, 2024-09-26:
Is this still happening? If so, where? 

I have tried to reproduce this using spread in the beginning levels but it didn't happen.