MiSTer-devel/SNES_MiSTer #351: SA-1 RAM write protection (Contra SA-1)

paulb-nl, 2022-11-06:
Contra III SA-1 v1.1 does not boot on a real SA-1 cart because RAM write protection is enabled by default.  This version does work on the SNES core but IMO it should behave like a real cart.

Contra III SA-1 v1.2 fixes this by disabling write protection:
https://github.com/VitorVilela7/SA1-Root/releases/tag/v1.8
https://github.com/VitorVilela7/SA1-Root/commit/8ec85aa8725c3600cb69e3000486009e23d32cf3
https://github.com/VitorVilela7/SA1-Root/commit/b67c8cc2aa56119b99cfc52789695f311bb60cdf

> This version fixes an issue regarding on how I-RAM (SA-1 internal RAM) is set up on boot code. At boot time, the I-RAM write stays disabled and you must enable it before making writes. However, my boot time attempts setting up the data bank register (DBR) that requires a stack operation (PHA - PLB) and the PHA opcode requires writing to I-RAM. Because write is disabled, the write is ignored and the DBR receives an undefined value instead (likely static randomic), which made some games not boot up at all. Contra III in particular never booted up on hardware because it requires the DBR to be 0x05 since the beginning, while Gradius III and Super R-Type had mixed results on hardware.
> 
> Now that the root cause has been found and fixed, Contra III, Gradius III and Super R-Type should work as expected on real hardware.

The SA-1 documentation in book2.pdf states that all RAM write protection is enabled by default: `SBWE`, `CBWE`, `BWPA`, `SIWP` & `CIWP`.

paulb-nl, 2024-10-08:
@srg320 Thanks for working on this!

Here is more information about BW-RAM protection and Kirby Dreamland:
https://github.com/ares-emulator/ares/commit/70f361094b2983bc783957083be9d7d46cf39bf2

Apparently both SNES and SA-1 write enable registers need to be 0 for BW-RAM protection to work. So because it sets `SWEN` to 1 both SNES CPU and SA-1 can write to BW-RAM.

Protection Test ROM:
https://github.com/absindx/SNES-TestRoms/tree/main/SA1RamProtectionTest

srg320, 2024-10-08:
> Here is more information about BW-RAM protection and Kirby Dreamland: [ares-emulator/ares@70f3610](https://github.com/ares-emulator/ares/commit/70f361094b2983bc783957083be9d7d46cf39bf2)
> 
> Apparently both SNES and SA-1 write enable registers need to be 0 for BW-RAM protection to work. So because it sets `SWEN` to 1 both SNES CPU and SA-1 can write to BW-RAM.
> 
> Protection Test ROM: https://github.com/absindx/SNES-TestRoms/tree/main/SA1RamProtectionTest

Thank you, this information is very helpful.

paulb-nl, 2025-01-19:
Testing with release 20241228. The SA1 Protection test now fails at test ID 155 which is `TestID_SNES_BwRamMirror_400000`

https://github.com/absindx/SNES-TestRoms/blob/2de2a7a565cb5aa2b361dbaa10c308e53f6ec0e8/SA1RamProtectionTest/MessageID.asm#L203

It says something in the readme about unmapped memory for test 145-162.

> TEST ID : 145-162
It seems that the unmapped area on the SA-1 side returns the old data bus value. (similar to PPU open bus...?)
(Probably) The initial value is an undefined value for a cold boot, and the previous value for a momentary power off or reset button.
Macro %SetSa1Databus() sets the bus to any value.
Since the details are still unknown, I plan to create another test ROM.

paulb-nl, 2025-01-22:
Great work! With commit d87e48c49c4be163b43fe2387ac0b5fe780fcc8e it now passes all the tests.

closed 2025-01-22 by paulb-nl