MiSTer-devel/SNES_MiSTer #352: Special Chip SRAM Cases

retrorepair, 2022-12-29:
There are a selection of special chip games that use a portion of cartridge SRAM as WRAM.

Currently the save file for these games will include a snapshot of this useless data but I feel it should only include the SRAM data.

The games with checksums and address vectors are as follows (from the SD2SNES source as this does handle these):

`// GSU
  { 0x5cb1755a, 0x7c00, 0x0400 }, // yoshi's island (us)
  { 0x42115ad4, 0x7c00, 0x0400 }, // yoshi's island (us) 1.1
  { 0x08f007cc, 0x7c00, 0x0400 }, // yoshi's island (eu)
  { 0x8a699987, 0x7c00, 0x0400 }, // yoshi's island (eu) 1.1
  { 0xb9b78a85, 0x7c00, 0x0400 }, // yoshi's island (jp)
  { 0x82efeef5, 0x7c00, 0x0400 }, // yoshi's island (jp) 1.1
  { 0x7c8fb8d3, 0x7c00, 0x0400 }, // yoshi's island (jp) 1.2

  // SA1
  { 0x0acd464f, 0x0000, 0x2000 }, // super mario rpg (us)
  { 0x44604774, 0x0000, 0x2000 }, // super mario rpg (jp)

  { 0x9897b7b6, 0x1F00, 0x0100 }, // kirby super star (us)
  { 0xdf749c91, 0x1F00, 0x0100 }, // kirby super star (eu)
  { 0x045c941a, 0x1F00, 0x0100 }, // hoshi no kirby super deluxe (jp)

  { 0xfdcd089c, 0x5E00, 0x0200 }, // kirby's dreamland (us)
  { 0x52707a84, 0x5E00, 0x0200 }, // hoshi no kirby 3 (jp)

  { 0xb7ad7461, 0x0100, 0x0C00 }, // marvelous (jp)
  { 0xb803d023, 0x0100, 0x0C00 }, // marvelous 1.06 (us)
  { 0x7a76f989, 0x0100, 0x0C00 }, // marvelous 1.07 (us Tashi-DackR)
  { 0x186dddd3, 0x0100, 0x0C00 }  // marvelous 1.07 (us DackR)`

[Source](https://github.com/mrehkopf/sd2snes/blob/develop/src/snes.c)

retrorepair, 2022-12-29:
I don't know if a name match may work better than a checksum to capture romhacks etc? I was looking into this but having a hard time wrapping my head around VHDL

sorgelig, 2022-12-30:
As long as save is included in save, it's considered very minor issue (if it's considered as issue at all).

paulb-nl, 2022-12-30:
The original cartridges contained a battery with a single SRAM chip so the entire SRAM content is preserved including the "useless data". There is no reason to change this behavior when the original cart is doing it.

Maybe the SD2SNES is just using those addresses to detect when to save SRAM to sd card.

retrorepair, 2022-12-30:
> The original cartridges contained a battery with a single SRAM chip so the entire SRAM content is preserved including the "useless data". There is no reason to change this behavior when the original [...]

Yes it does, I had the idea MiSTer could eventually do the same but I think there's other ways round it.

closed 2022-12-30 by retrorepair