MiSTer-devel/SNES_MiSTer #354: Treasure Conflix (Bandai Satellaview) doesn't display correctly either text or graphics

ValcristRoda, 2023-01-20:
A BS-X game got translated to English recently called Treasure Conflix and I decided to test it out. The text is unreadable and when you get to the actual game all the sprites are either more blocks or cut out black areas on the screen. Tested a non-patched Japanese rom and it was the same way.

birdybro, 2023-01-20:
I'm using this rom [which matches the no-intro entry](https://datomatic.nointro.org/index.php?page=show_record&s=77&n=3122):

MD5: 534DE4685ED6A099A31717826AD8AD8B
SHA-1: DFF1D22D662FED96F45AC3E722C79C93469CA421
SHA-256: E13DA225FB5B72F3B694D68A478A58A9F6E594A2ECB52E9C8CEFDEE0B60FD567

Text looks fine to me on the non-patched.

[image]

Sprites and text at the same time works fine.

[image]

mode 7 looks fine:

[image]

Make sure you are using the no-intro rom and not a hacked rom like what used to be required before BSX support was added.

jhdgghost, 2025-02-12:
I confirm this issue. I checked the checksum of the ROM and BSX BIOS, and both appear correct, but the graphical corruption persists, as previously reported by [ValcristRoda](https://github.com/ValcristRoda).

For reference, I am using a MiSTer Pi from RetroRemake and loading the ROM from the SD card provided with the Turbo Pack (Batch 2).

birdybro, 2025-02-12:
Can you post a picture of the kind of graphics corruption you are seeing?

jhdgghost, 2025-02-22:
Sorry for the delay. Here some videos the core i used is [SNES_unstable_20250213_19b75b.rbf] and was the same in last stable release.

translated rom
https://drive.google.com/file/d/1Zi2ibeLi1HPB_VTcljHc0EXrSv1uCbtz/view?usp=sharing
Untranslated rom
https://drive.google.com/file/d/1OHlIt_Or7OnsCL3Mkp699f8zhgCchfdg/view?usp=sharing

The same rom in RA with snes core. The rom is soft patched. Japanese ROM MD5: 534DE4685ED6A099A31717826AD8AD8B
https://drive.google.com/file/d/14EQS5DUJ79v2k5qe0oiQlF3suX6fwCs4/view?usp=sharing

Sorry,i capture with the phone but think the glitch are very evident. Text and mode 7 have corrupted graphics.

jhdgghost, 2025-03-06:
Ok, i resolve the mistery. All my BSX games have sfc or smc extension i changed to bs extension and that fix the issue. Now is loaded as BSX game with the bios before was launched as cartrige game.

That fix the problem. The header probably is no correct or the core was not able to detect the game properly without the  extension.

The problem can be closed.

closed 2025-03-07 by srg320