MiSTer-devel/SNES_MiSTer #355: No game runs?

jeffbdavenport, 2023-02-08:
Is it just me? I can't get any SNES game to run on any of the releases. Best I can get is some of them will get to the intro of the game and as soon as you click play it freezes. 

Yes I have the 32MB SDRAM. All NES and Genesis games run just fine.

birdybro, 2023-02-08:
Can you provide a picture of your SDRAM as installed?

Can you load games in the NES core?

jeffbdavenport, 2023-02-08:
[image]

Yes I can load the NES core as well as any games and they all run flawlessly

birdybro, 2023-02-09:
That SDRAM v3.1 is 5.5 years old. It's a collector's item! :)

My first inclination is to suggest that you upgrade to a modern SDRAM module since it's possible that this model isn't wired up identically and the SNES core might be taking advantage of a line that isn't present.

Something I noticed is that multiple pins do not have the proper amount of solder so you may have signal integrity issues or even a full disconnection if it gets even slightly budged.

[image]

jeffbdavenport, 2023-02-09:
Interesting, I thank you for your insight, I was suspicious that it was the Ram but was confused that memory tests ran just fine and other cores worked fine as well, I will try to get the latest SDRAM version

closed 2023-02-09 by jeffbdavenport