MiSTer-devel/SNES_MiSTer #356: Super Metroid MSU-1 Audio Crackling

buvk, 2023-02-25:
I am experience audio crackling when playing Super Metroid with MSU-1 (I tried a few different music packs).  This is especially noticeable in the menus (E.g. when you select START GAME and the map loads), and in-game when firing Samus's blaster.

I do not experience this if I remove the PCM files.  I also do not experience this in other MSU-1 games that I have tested.

I am using HDMI audio, but I also experience this using analog audio.

There does not appear to be any audio crackling when playing Super Metroid with MSU-1 in an emulator (tested using Snes9x).

sorgelig, 2023-02-26:
probably it exceeds the speed core supports.

birdybro, 2023-02-27:
Try and disable this option if it's turned on --> https://github.com/MiSTer-devel/Main_MiSTer/blob/master/MiSTer.ini#L9

On some displays I've heard crackling when this is turned on. Some kind of compatibility issue.

buvk, 2023-02-28:
I tried enabling and disabling this setting, but it did not make a
difference.

paulb-nl, 2024-10-02:
I have looked into this and the MSU patch always writes $00 to $2140 to disable the original music.

On one of my SNES consoles this causes a crackling/pop sound at the end of some sounds. This is without MSU. The pops are caused by the SPC700.

I think what happens is that because the sound driver never plays any music, it never initializes some memory and now it relies on uninitialized memory values.

If you set "Initial ARAM" to 00FF or 55 then everything sounds fine without pops. Setting it to FF will produce the loudest pops similar to one of my SNES consoles.

paulb-nl, 2025-01-22:
Can be closed. It is a game patch issue with workaround available.