MiSTer-devel/SNES_MiSTer #357: Hook - fastrom hack visual glitch

leonquest, 2023-02-27:
Played this fastrom hack on the latest version, and noticed graphical glitches on the story sequence right after the first battle with capt hook.

Wrote the hack owner and he mentions could not recreate it on snes9x.

I'm looking to recreate it on real hardware, but haven't found working cheats for level select.

leonquest, 2023-02-27:
> Snes9x specifically has many inaccuracies when compared to original hardware. Can you test with Higan instead and report back?

I need to find usable cheat codes to test on real hardware, but haven't found one yet.

birdybro, 2023-02-27:
So it doesn't run on Higan without cheat codes? I'm surprised it runs at all on the MiSTer core then.

leonquest, 2023-02-27:
* I have not tested on higan and not planning to, not sure where you read that in my comments

* the cheats I mentioned are for warping into the area where the glitches were seen, I don't have time nor the will to go through the game again, even if its a short playthrough, so I'm looking for another way to fast forward to the location.

birdybro, 2023-02-27:
Ah got it, I requested that you test it in Higan, hence my confusion toward you response.

Why are you not willing to test with Higan? snes9x is known to be inaccurate when compared to original hardware, as compared to Higan. snes9x has compatibility with specific methods of hacks that will only work on snes9x as well. The MiSTer core uses much of the documentation from no$ and from bsnes (now Higan) for it's design and is made to be very accurate to original hardware. Therefore, snes9x shouldn't really be used as a base of comparison for what the MiSTer core ought to do.

leonquest, 2023-02-27:
Please read my first post carefully

birdybro, 2023-02-27:
I did. You played a fastrom hack, it got graphical glitches on a later level in the game, and you mentioned the author of the hack couldn't recreate the bug in snes9x, which is an inaccurate emulator that caters to behavior that isn't present on real hardware, by design, for hacks.

leonquest, 2023-02-27:
Tested on a 1CHIP Fat snes with an SD2Snes rev H, no glitching after the first battle with hook. I used all kinds of cheats to get there in half an hour, as unfortunately there are no GG codes out there yet for level skip or level select, and the supposed code left in the game by devs does not work.

Now, to fully confirm can someone else test again on a mister? I might find time again in the near future but maybe not today.

Word of advice if you use cheats to recreate quickly, turn them off right after finishing each stage otherwise the game will score you infinitely until it is able to drain your time, health and flying meter.

birdybro, 2023-02-28:
Thanks for the thorough testing.

leonquest, 2023-02-28:
Thanks to the cheats I was able to re-test relatively quick on mister.

Tested both a clean no-intro version and the fastrom hack. Could not reproduce it 🤷. I'm not sure what happened the first time, maybe it was a one-off or maybe it's part of the game's code.

Will shamefully close this one out.

closed 2023-02-28 by leonquest

birdybro, 2023-02-28:
No shame at all! :)