MiSTer-devel/SNES_MiSTer #365: SNES Mario Paint: original mouse delay and stop when rubber is activated

superbonaci, 2023-04-23:
The issues are mentioned in this video (from 13:35 to 14:00):
https://www.youtube.com/watch?v=AyDqGdY7ffQ

- The original Nintendo mouse with mousepad has certain delay
- Clicking when moving the eraser stops the movement

The youtuber claims such things don't happen on original console.

bikerspade, 2024-09-25:
Direct link to 13:35: https://youtu.be/AyDqGdY7ffQ&t=815
Have you tried this on original hardware?

superbonaci, 2024-09-25:
> Direct link to 13:35: https://youtu.be/AyDqGdY7ffQ&t=815 Have you tried this on original hardware?

No I don't.

bikerspade, 2024-09-25:
I just tried this on real hardware and a Super Nt with an OEM mouse and the behavior is identical with the eraser. This issue can be closed.

superbonaci, 2024-09-26:
Closing then.

closed 2024-09-26 by superbonaci