MiSTer-devel/SNES_MiSTer #371: Sync issue with snes/sgb cores ntsc and LG Flatron 21FB crt - Wobby picture

stainmage, 2023-07-16:
Hello,

I have specific sync issue with this CRT, it doesn't appear with other tv sets, nor with other cores with the same tv.

Waves go vertically acrosse the screen, wobbly picture. I don't understand why but it seems this tv is picky with this core only in NTSC mode.

I'm using Antonio Villena's scart with Mister Plus

Cheers

birdybro, 2023-07-18:
Did you email @antoniovillena for support yet?

info@antoniovillena.es

He might know how to resolve this more easily than most.

stainmage, 2023-10-07:
yes made an email, however we didn't find the culprit yet, I think it's a incompatibility between the sync signal of the snes core and the tv, still couldn't test with another cable but I'll do it later when I will reach the tv again

birdybro, 2025-08-06:
What's the make and model of your CRT?

stainmage, 2025-08-06:
LG Flatron RT-21FB75VX

stainmage, 2025-08-06:
yeah I was suspecting that but why would it happen only with this core on ntsc ? all other cores I have tested on ntsc don't give this beahviour