MiSTer-devel/SNES_MiSTer #373: Kaizo Hack Passcode Mechanism not working

gizmo-321, 2024-01-19:
Hi there,

I tried to play the SMW-Hack Chain Reaction (https://smwc.me/s/18669) and the Creator uses a Passcode-System to go to a Level Selection. You have to jump on each box to count them up. But this doesn't work. The Box stays at zero. I jumped also on them and count by myself, but this also doesn't work. Maybe it's possible that your core is missing a feature to be able to work with this passcode system, but I don't know technical details how this Passcode System works.

Same Rom work fine on the SuperNT and FXPakPro...

This problem was at first at my Analogue pocket. In the meantime I bought myself a MISTer and the problem is there also present. I opened the issue at first at agg23's githug page, and he meant I should post it here. Also in the meantime I found out that you have to jump over 100 times on each box and then the box begins to count ;-)

[image]

paulb-nl, 2024-01-20:
The developer of this hack did not clear RAM before using it so the results are depending of the initial RAM state of the SNES console.

If you change the setting Hardware->Initial WRAM to 55(SD2SNES) then it will work better. (Need to jump twice to increment from 0->1). It looks like the developer expected RAM to be all zeros which is not always the case on a real console.

The Super Everdrive and FXPak both appear to be filling RAM with 55 before starting the game.

gizmo-321, 2024-01-20:
Thanks, this also works for me :-)
Jump twice was also necessary on the FX Pak Pro

gizmo-321, 2024-01-20:
Thanks, this also works for me :-)
Jump twice was also necessary on the FX Pak Pro

So it's not a core problem, just a Hack and Settings-Problem
Maybe also less knowledge in this case ;-)

closed 2024-01-20 by gizmo-321