MiSTer-devel/SNES_MiSTer #374: Freezes at certain stages.

Vashfairytail900, 2024-01-23:
Downloadable from View in Files on this page
smwx.bps in hackrom
DONUT PLAINS 4 (course #6), STAR WORLD 5 (course #136), FOREST OF ILLUSION 1 (course #11E)
The game freezes when I launch each of the following stages of the game
There are no special chips or other settings.
This is a hackromes created by patching to the original Super Mario World.
It worked fine on other emulators (BSNES, Snes9x etc) and on the actual SNES machine with SF memory cassette.

https://www.smwcentral.net/?p=profile&id=29537

Vashfairytail900, 2024-01-23:
PS.
The game has a debug patch that allows this hackromb to travel freely to and from the Overworld.
The closest to the relevant part is the course just before the castle on the Donut Plain.

birdybro, 2024-01-23:
I don't see a link to any patch for this hack you are referring to.

Vashfairytail900, 2024-01-24:
Did you select View from the Files column?
The smwx.bps file in the View link is the patch for the relevant hackrom.

MP2E, 2024-01-24:
https://bin.smwcentral.net/u/29537/smwx.bps

here's a direct link

I seem to have the same issue, I tried changing the Hardware memory settings to SD2SNES but still encountered the crash

paulb-nl, 2024-01-24:
I have tried Donut plains 4 and it freezes in Bsnes, Ares and Mesen2.

birdybro, 2024-01-24:
Ah I see now, sorry for the confusion. It wasn't very intuitive.

[image]

MP2E, 2024-02-11:
> I have tried Donut plains 4 and it freezes in Bsnes, Ares and Mesen2.

Just tested this on my 1chip SNES with an sd2snes and it also crashes just like the aforementioned emulators.

I think it’s safe to say this is a hack issue, not an issue with the core.

Vashfairytail900, 2024-02-12:
The details of the problem are now known.
It appears that it was a RAM conflict.
The fixed version has suppressed the crashes.

sorgelig, 2024-02-12:
So it's fixed and can be closed?

paulb-nl, 2024-02-12:
Can be closed. It is not an issue of this core.

closed 2024-02-12 by sorgelig