MiSTer-devel/SNES_MiSTer #376: MSU-1 game audio isn't working properly in the 20240408 build

idelacio, 2024-04-27:
MSU-1 game audio isn't working properly in the 20240408 build.

FMV cutsecenes seem to load fine but the sound effects and music are distorted, especially on the NES ported games. On loading a regular SNES rom the issues follow until the system is reset.

I reverted to the 20231204 build and the sound was completely fine.

Tried with the MiSTer_20240110 and MiSTer_unstable_20240331_04a227 builds of MiSTer, issue persists across both.

My MiSTer ini and SNES config-
[settings.zip](https://github.com/MiSTer-devel/SNES_MiSTer/files/15137798/settings.zip)

sorgelig, 2024-04-27:
I've tested on "Super road blaster (MSU1)" and didn't find issues with sampled music in intro.
How you get audio from MiSTer?

idelacio, 2024-04-27:
I've updated the ticket title to better reflect what I think the issue is... >.<

Looks like I let myself get tricked... I THINK what's happening is that some of the 2D elements are possibly what's causing the music to wig out. Of course some of the games I tested after trying MSU1 games were of course games with just music and simple BG layers in the intro, so that threw me. Super Mario World turns out to be a good test for this.

At first I thought it was sound effects and music together causing the issue BUT I noticed that MSU1 and regular Doom are apparantly fine, the Williams intro is clearly badly crunchy but after that it sounds okay. As in when there are no traditional sprites on screen. Tried StarFox and StarFox 2 and seem to be much better when there are no 2D elements on screen. Stunt Race FX is pretty bad but it has a large HUD over it.

Knowing it's just me, tested further (and in answer to your question!)-
I have two MiSTer Ironclads of different revisions.
One has the audio expansion board and is the one with the issue.
The other one does not and is fine on analogue and digital.
They are using the same settings and versions. 
So the only meaningful difference is that the SW0 switch on the DE10 has been toggled on the one with the issue.
That's disabling the audio generator on the DE10. Odd that it only affects this revision of the core and only the MSU1 sound though, but the framework was recently updated?

So far I haven't seen this issue in any of other cores (yet). I haven't seen any issues in the Megadrive, NES, or Mega-CD cores that all recieved a framework update at the same time (inc MUS-MD games).

sorgelig, 2024-04-27:
I don't know what is "MiSTer Ironclad". If it some HW addons, then you need to consult with maker of that board. I support only official addons (released by me).

idelacio, 2024-04-28:
It's a rejig of the Mister expansion boards into an mini ATX design.

No worries, I'll drop them a message. As the previous version of the core works without issues I'll leave it at that.

closed 2024-04-28 by idelacio