MiSTer-devel/SNES_MiSTer #377: Super Metroid + Link to the Past + Metroid + Legend of Zelda Quad Randomizer - Metroid Transition Does Not Load

TooOrange, 2024-08-09:
I don't know if this is something that would be addressed or even be possible to fix, but I just wanted to report that the new randomizer combining Super Metroid, Link to the Past, Metroid, and the Legend of Zelda is not fully functional in this core. The game does properly load up and play, however attempting to transition into Metroid simply causes a black screen. All other game transitions and functions of the other three games appear to be as expected. The game does work without issue on original hardware using an FXPak or SD2SNES, as well as in BSNES, though newer versions of SNES9X after 1.58 will not load the game at all. I suspect it may be an issue with how the core handles SA-1, similarly to the problem with the Link to the Past practice ROM mentioned in issue [#343 ](https://github.com/MiSTer-devel/SNES_MiSTer/issues/343). However, my knowledge of this is very limited. Not necessarily expecting any kind of possible fix or anything, but just mostly wanted to make it known for anyone else that may try to use the core for this.

steven-a-wilson, 2024-08-24:
I also just hit this and came to report it. I looked into it a little bit, but I'm also not enough of an expert on SA-1 or the SNES core's code to have a really solid grasp of what's going on. The hack seemingly uses the SA-1 mostly as a memory mapper. Looking at the Mesen debugger, I see the transition to M1 making these changes to the mapper registers:

$2220: $84 -> $80
$2221: $85 -> $80
$2222: $84 -> $86
$2223: $85 -> $07
$2224: $01 -> $04

I don't see anything here that seems obviously broken or missing in the current code, though it's hard to compare to software emulators because they're structured so differently. Perhaps there's a bad interaction between the SA-1 mapper logic and the 64-megabit ROM mapping? I can't imagine there are very many 64Mbit SA-1 games out there for the core to have been tested against.

sorgelig, 2024-08-26:
I think it's the same "old" issue when hacks use out of specs SA-1 access. This core already has tight timings barely fitting standard SA-1 access.

paulb-nl, 2024-10-04:
The issue here is that the Metroid port attempts to read from NES register $2002. This register is open bus on a SNES.

LDA $2002 is $AD 02 20 in bytes so reading $2002 should read as $20 from the last read value (open bus) but the SNES core reads $C5 and it never breaks out of the loop.
[image]
[image]

srg320, 2024-10-05:
Where can I get the rom?

paulb-nl, 2024-10-05:
You can generate the rom here https://quad.beta.samus.link/

For Zelda it needs the japanese Link to the past and NES Zelda PRG 0

Here is a video how to transition to Metroid.
https://github.com/user-attachments/assets/daf658f7-a5e1-4b3b-8bf9-9fb6041f39a4

steven-a-wilson, 2024-10-06:
For what it's worth, the Metroid mode works if all instances of that loop are patched to replace the read with a constant load of $20, so this appears to be the only thing stopping it from working. This was a very quick-and-dirty test; all I did was load it in a hex editor and do a search and replace of:

`AD 02 20 29 80 D0 F9`

with

`EA A9 20 29 80 D0 F9`

paulb-nl, 2025-05-29:
That is another issue. There is a fix available that should make that hack work correctly.
https://forum.metroidconstruction.com/index.php/topic,5443.msg70105.html#msg70105

paulb-nl, 2025-05-29:
This issue can be closed. Fixed in release 20241228.