MiSTer-devel/SNES_MiSTer #381: Kirby’s Dreamland 3 freezes before first level

MP2E, 2024-09-25:
This issue was reported by zazaza on the forums, KDL3 (JP or US) freezes on the latest release version but does not freeze on the previous release. 

Thanks!

italiangrandma, 2024-09-25:
I also tested this with both the JP and USA versions on the 09192024 release of the core and cannot reproduce the issue.

Zet-sensei, 2024-09-25:
Can't reproduce with the 2024-09-25 version either.

I know that some SNES core past revision had some errors with SA1 games -Like with Kirby 3, Mario RPG and Jikkyou Parodius- (tested on 2024-09-18 & reported on Discord).
As it was an incorrect build done automatically from the bot.

mambrino, 2024-09-25:
I've updated and download manually again v240919 version and I still have the same issue. The game freezes before first level.
Tested with USA (CRC32: ec8a48f6) and Japan (CRC32: 43373aee) no-intro roms.
Deleted snes.cfg and same behavior again.
I have also tried with the unstable version (v240925) and the game doesn't boot, all I have is a black screen.
My mister is a standard DE-10 Nano and never had problems with snes core until now.
Maybe is a ram issue? (mine is sdram 128mb v2.9)

wwark, 2024-09-25:
Have you issue with all SA-1 games ?
Could you please try this test rom ?

https://github.com/VitorVilela7/SnesSpeedTest/blob/master/speed_test_v51.sfc
and share a screenshot if the rom works.

littlegamer87, 2024-09-25:
Just installed rom w/CRC ec8a48f6 and beat the first level with no issue. 
With issues like these, I would suggest to reinstall your sd with mr fusion, update and try again. If that doesn't work, likely hardware failure.

ZZZyeahZZZ, 2024-09-25:
> Have you issue with all SA-1 games ? Could you please try this test rom ?
> 
> https://github.com/VitorVilela7/SnesSpeedTest/blob/master/speed_test_v51.sfc and share a screenshot if the rom works.

@wwark - speed_test_v51.sfc works on SNES_20240919.rbf and SNES_20240408.rbf.

took screenshots of pages 1-4 on both cores, for a total of 8 screenshots. let me know of any particular page(s) and which core would be useful to upload.

wwark, 2024-09-25:
@ZZZyeahZZZ  all of them and separated them for each version.
It could help to see there is any differences on SA-1 test.
Thank you for your time and for the test

ZZZyeahZZZ, 2024-09-25:
@wwark here you go, and np, sincerely hope it helps.
[speed_test_v51.zip](https://github.com/user-attachments/files/17137447/speed_test_v51.zip)

littlegamer87, 2024-09-25:
I checked the speedtest with my results and found nothing. Curious it runs for impacted users. The only real difference between said core versions is WRAM moved to SDRAM (to make space for BSRAM), which leads to believe there is something different about SDRAM for users impacted. I invite these users to come up with ideas what that could be. Try other SA1/non-SA-1 games. Do a ram test. Anything that can help.

Zet-sensei, 2024-09-25:
I can share some measurements done with the 2024-09-25 build and all SA1 games are working on my systems.
For reference & possible comparisons.

[image]
[image]
[image]
[image]

mambrino, 2024-09-25:
here are mine. 240408 works perfectly, but 240919 freezes regardless of rom or config. Tested other SHA1 games and super mario rpg freezes too, but jikkyou oshaberi parodius works perfectly for me.

Tested my sdram (v2.9 single) and is stable for 10min at 140mhz.

[SNES ram test.zip](https://github.com/user-attachments/files/17138697/SNES.ram.test.zip)

ZZZyeahZZZ, 2024-09-25:
@mambrino  jikkyou oshaberi parodius is also functional for me on 240919. only played it for a few minutes, not sure if it crashes eventually. 

@wwark i'm on mister v240912

MP2E, 2024-09-25:
I just tested this on my MiSTer and I’m not getting the crash either.

The reason I opened an issue is because I saw 2 different people report this on the forums.

One thing that might be worth a try for anyone having the issue is to download the SNES MiSTer Nightly build and try that.

wwark, 2024-09-26:
hi @MP2E 
Could you please reopen this issue because there is other cases also on discord.
Maybe it could be link to change on SDRam side and need to be understand.

Thanks !

MP2E, 2024-09-26:
Yes no problem, I have reopened the issue.

ZZZyeahZZZ, 2024-09-27:
Update: swapped my v2.9 ram with v3.0 ram (both have the Alexy Melnikov copyright, both were purchased from Misteraddons), problem persists. i switched back to v2.9

ran memtest on v2.9 ram. 10 minutes passed, stable at 145 mhz

rhester72, 2024-09-27:
Folks who are having trouble: Please provide the md5sum of both your MiSTer main binary and the SNES core.

javidoom, 2024-09-28:
Hi,

I also have this problem, using the same snes core version (19/09/24), all sa1 games freeze before first level or during gameplay.
i'm on mister version 12/09/24 and Kirby Dreamland 3 - CRC32: EC8A48F6
I've downloaded 19/09/24 version again (manually) and I still have the same problem.
I have also tried with the unstable version and all sa1 games freeze with black screen.
My mister is a DE-10 Nano very old one from many years ago and never had problems with other cores. It's not a clone.
My SRAM is an official v2.4 single ram stable at 140mhz in ram test.

szuping, 2024-09-28:
Hello, SNES_20240925.rbf can hardly run any SA-1 games (JAP or USA) on my MISTER, but SNES_20240408.rbf is normal.

ZZZyeahZZZ, 2024-09-29:
> Folks who are having trouble: Please provide the md5sum of both your MiSTer main binary and the SNES core.

mister main v240912  md5: 836457367d3ddb28b7b59ade107cffb7
SNES_20240919.rbf md5: cebe1ce14259fba339ed4421d5c18e21

mambrino, 2024-09-29:
> Folks who are having trouble: Please provide the md5sum of both your MiSTer main binary and the SNES core.

mister main v240912 md5: 836457367d3ddb28b7b59ade107cffb7
SNES_20240919.rbf md5:  cebe1ce14259fba339ed4421d5c18e21

italiangrandma, 2024-10-01:
For everyone affected, please test again with the latest unstable core build (10012024)

ZZZyeahZZZ, 2024-10-02:
thank you @italiangrandma .

i've tested https://github.com/MiSTer-unstable-nightlies/SNES_MiSTer/releases/download/unstable-builds/SNES_unstable_20241001_16fe7f.rbf and can report the following: 

KDL 3 working (first level loads, i can play it)
kirby super star playable
super mario rpg running at normal speed  
dragon ball z hyper dimension has no graphics corruption
super bomber man panic W working

@mambrino hopefully the unstable core above solves your issues as well!

szuping, 2024-10-02:
> 謝謝@italiangrandma。
> 
> 我已經測試過https://github.com/MiSTer-unstable-nightlies/SNES_MiSTer/releases/download/unstable-builds/SNES_unstable_20241001_16fe7f.rbf並且可以報告以下內容：
> 
> KDL 3 工作（第一級加載，我可以玩） 卡比超級明星可玩 超級馬裡奧角色扮演遊戲以正常速度運行 龍珠 Z 超維度沒有圖形損壞 超級轟炸機人恐慌 W 工作
> 
> @mambrino希望上面的不穩定核心也能解決您的問題！

MiSTer_20240916
SNES v20241001
[VitorVilela7](https://github.com/VitorVilela7)
The SA-1 game patched games here can also be started normally, and the graphics are no longer damaged and frozen.

MP2E, 2024-10-02:
Sounds like this issue can be closed!

closed 2024-10-02 by MP2E