MiSTer-devel/SNES_MiSTer #382: CPU: Turbo setting not working on latest build (09252024)

italiangrandma, 2024-09-26:
Setting the CPU Setting to Turbo causes any game I load to immediately crash to a black screen or garbled graphics. 

Potentially related to #381 ?

ZZZyeahZZZ, 2024-09-26:
@italiangrandma  Thanks for reporting this. Tested first with Gradius 3, then with more games:

Same for me, CPU set to "turbo" in the SNES core crashes in SNES_20240919.rbf.

"Turbo" in SNES_20240408.rbf is functional.

sorgelig, 2024-09-27:
@srg320 
i believe it's related to a new SDRAM controller.

wwark, 2024-10-01:
It seems fixed no more issue on my side !

italiangrandma, 2024-10-01:
yep, appears to be fixed with ltest unstable build (10012024)!

closed 2024-10-01 by italiangrandma

ZZZyeahZZZ, 2024-10-02:
I've tested https://github.com/MiSTer-unstable-nightlies/SNES_MiSTer/releases/download/unstable-builds/SNES_unstable_20241001_16fe7f.rbf , and it appears the "turbo" CPU setting is functional (at least with gradius 3).