MiSTer-devel/SNES_MiSTer #384: Graphic corruption on Super Mario RPG opening

ctwohig, 2024-10-19:
When loading the opening sequence in Super Mario RPG i see graphic corruption in the butterflies and a few other items.  The game load screen also shows problems feeling sluggish trying to move between saves.  I have tried the retail release and virtual console file from the No-Intro pack, neither file shows the problem in RetroArch with the bsnes core and match the expected No-Intro checksum.  I'm guessing there are more issues i just haven't played through any further after encountering this issue

Zet-sensei, 2024-10-20:
There was some issue with the SA1 chip some time ago (like Super Mario RPG) & srg320 updated the code on the last unstable version.

Fixed with that commit https://github.com/MiSTer-devel/SNES_MiSTer/commit/ad3ae5e746647880119efd33ee9401c2a2a1306d

Please retest with a more recent (unstable) version.

Jromr, 2024-10-21:
I had the same issue on the stable core, can confirm this is fixed on the latest unstable Snes core on the discord uploaded on 10/7 (20241007_11ef03.rbf)

Zet-sensei, 2025-01-13:
@ctwohig Could you please restest with the last stable version of the core and verify everything is okay now.
If so, could you please close that issue.

Thanks.

closed 2025-06-29 by srg320