MiSTer-devel/SNES_MiSTer #386: Please add an overscan on/off option

kandowontu, 2024-10-22:
SNES has overscan features that would be amazing to be able to toggle on and off for certain games.

Hopefully adding this feature will be a trivial task!

birdybro, 2024-10-22:
Which games specifically would benefit from this?

kandowontu, 2024-10-22:
> Which games specifically would benefit from this?

You can find a list here:
https://snes.nesdev.org/wiki/Uncommon_graphics_mode_games#Overscan_(239_lines)

This is also for future ports and romhack projects to be able to make use of this area.

birdybro, 2024-10-24:
Do you have examples where the MiSTer core is missing something versus bsnes for example? I tested Super Tetris 3:

bsnes:
[image]

MiSTer core:
[image]

I don't think I see any difference between to two (despite my filters being different and MiSTer defaulting to DAR more accurately).

kandowontu, 2024-10-24:
After reviewing it, it appears that MiSTer ALWAYS displays the overscan area.

This has been troublesome for some NES->SNES port developers who have relied on the overscan area to place graphics for use in some cases.

It may also be useful for getting better full picture by removing the overscan on non-overscan games.

paulb-nl, 2024-10-24:
Toggling overscan is done by software so I don't understand what you mean by "troublesome".

"Getting better full picture" is not possible because that means Vblank time is shorter and games will show glitched graphics.

kandowontu, 2024-10-24:
Understandable. I'll close the issue and port developers will just have to make sure to disable the overscan register for MiSTer compatibility.

closed 2024-10-24 by kandowontu