MiSTer-devel/SNES_MiSTer #387: Brandish (USA) // Incorrect track played for some secdonds during the ending credits

Zet-sensei, 2024-11-05:
I'm opening that issue for the account of a user (Xpert74) on Discord.

_I beat Brandish in the SNES core (dated 2024-09-19) recently... and I think I encountered an emulation error here, at the end._

https://youtu.be/AOE-HjVC0t8?si=eab9uMWe1ojz5-k8&t=254

_Specifically - when you beat the game, and the ending story sequence has played out, you then have the credits sequence begin (4:14), while flashbacks to previous boss encounters in the game happen. However, there's an issue where the ninja boss fight's music track overrides the credits theme (4:19), and said boss theme keeps playing up until the game loops back to the start-up sequence._

https://youtu.be/QxtmxivtsN0?si=eEMZzZ7_Annjm-Kj&t=12254

_When comparing to this longplay, the credits theme here does not get overridden by the boss theme, and the credits music is thus able to play out in full._

_I think this is likely an inaccuracy with the Super NES core itself._

_I should note that I played through the game using a couple romhacks that restore the Japanese art/music, as well as changes the character names to what they were in the Japanese version. I went and tested this again using just a standard unhacked ROM, and the same music error took place._

_This is my save file, if anyone else wants to test it and see if they get the same music error as I do._

[Brandish.zip](https://github.com/user-attachments/files/17632612/Brandish.zip)

Zet-sensei, 2024-11-05:
After a retest requested from my part with the current core (10-30-2024).

_I retested it using the unstable 2024-10-30 core._

_The same music glitch persists, in the same way as before._

https://youtu.be/GIpdyokCEAA?si=Pt1j0DG0rm37fUXf&t=282

littlegamer87, 2024-11-06:
I've observed that it is not just the music that doesn't match. The mister core appears to also play weapon sound effects, the trap door sound and that three headed dragon attack sounds, and the long play video does not. This detail is interesting because it suggests that the code sent to the SPC700 for the sound effects is present at that scene in the rom when it is not supposed to play, and that might be true for the music too. My guess is the game is doing something unintended with the SPC700 and the response is not the same.

littlegamer87, 2025-05-31:
In an attempt to contribute and learn some new skills, I spent some effort on debugging this with Bizhawk BSNES+ core. This core does the exact same thing as mister wrt the music, so it looks like OP has found unintended SPC700 behavior in this game that is not properly implemented in emulators either.

The game is definitely writing to the SPC700 after the correct song has loaded, so I still believe the game is crashing the SPC700 so it won't accept incoming data anymore and loop the song endlessly. After that it looks like it reuses code which also loads new music. Bizhawk and mister both execute this music, real hardware does not.

The first breakpoint (at SPC700 writes) I found after loading the correct music is at PC `00D40E5A`. I found suspicious activity after it branches to PC `00D40E12` where it sends the command `04 65 8E` to the SPC700 around the time of the crash. This command appears only once in the trace.

I think that's all the progress I can make for today. In the meantime, if anyone can confirm if the instruction `04 65 8E` is illegal and/or ignored by the core, then we might be on to something.

srg320, 2025-05-31:
@littlegamer87 
I can't try to find the problem because I can't beat the final boss). 
Is there a way to make a save in the BSNES after beating the boss and before the final scene starts?

littlegamer87, 2025-06-20:
Many thanks srg! I will now try to find that flag.

I think we can conclude that this is not a core bug, this is a game mechanic (or a game bug, behavior is still weird) that reporter was not aware of. @Zet-sensei , please close. Thanks!

closed 2025-06-20 by Zet-sensei