MiSTer-devel/SNES_MiSTer #389: Culture Brain titles graphic glitch with analogue IO

xenyou, 2024-11-21:
It seems that some of Culture Brain titles have a graphic glitch with analogue IO board.

The titles I checked are:
- SD Hiryuu no Ken (Japan)
- Super Chinese World (Japan)
- Super Chinese World2 (Japan)
- Super Chinese Fighter (Japan)

Only audio is fine. But the graphic is flickering and I can't play it.
It's also reproducible with HDM and DAC (direct_video=1). 

It might be fine with CRT display. I'm using LCD monitor (15khz supported).

birdybro, 2024-11-22:
Cute bird.

Have you reset the core settings to default to see if the problem goes away?

xenyou, 2024-11-22:
> Have you reset the core settings to default to see if the problem goes away?

I've tried it. But it won't work.

And, I found this title is also reproducible.
- Super Ultra Baseball 2 (Japan)

paulb-nl, 2025-01-27:
What does the graphic glitch look like?

xenyou, 2025-01-31:
I tried it again today. But it works!!

Hmm...
VGA cable looks deteriorated.  So it might be cause.

I'll close this issue.

closed 2025-01-31 by xenyou