MiSTer-devel/SNES_MiSTer #391: Is 128kb vram supported by the ppu

slidelljohn, 2024-12-27:
Is 128kb vram supported by the snes cores ppu? Just curious because the original ppu in the snes does support the extra vram. I created a mod chip for the original 2 chip snes motherboards to test the 128kb vram and it works perfectly on all original 2 chip motherboards hardware. Having support for this on fpga would give others access to the extra vram and it could lead to some interesting projects.

“Now we’re playing with power, super video ram power!”

paulb-nl, 2024-12-30:
Probably yes but a mod is required for the SNES core too.

This is not something that can be added to the normal core because the FPGA ram is already fully used so it will require a separate core.

What software did you use to test 128KB VRAM?

srg320, 2024-12-30:
As a variant, we can take memory from the backup sram. That is, if the "Vram128k" option is enabled, the backup will be limited to 128kb instead of 256kb. But is there any sense in this option?

slidelljohn, 2025-01-03:
> Probably yes but a mod is required for the SNES core too.
> 
> This is not something that can be added to the normal core because the FPGA ram is already fully used so it will require a separate core.
> 
> What software did you use to test 128KB VRAM?

I used higan v100 at first. Higan had a hidden feature for 128kb of vram that byuu added. I used a modified version of bsnes plus that I think is on GitHub. The modified bsnes plus not only uses the 128kb of vram but it also has another 128kb set of vram that you can switch to using the unused I/O ports on the cpu. This also works on original hardware with the modchip I made. I also used that same modified bsnes plus debugger plus I made some of my own modifications to the source so I could do some other tests.

> As a variant, we can take memory from the backup sram. That is, if the "Vram128k" option is enabled, the backup will be limited to 128kb instead of 256kb. But is there any sense in this option?

Yes, 128kb vram is definitely worth it. The 2 sets of 128kb of vram is questionable but I made use of that as well when I converted mega man x1 to be able to use 128kb vram x2. If I can get more people to have access to this then more stuff can be made and tested and if stuff is made I’m sure something like the super nt might consider adding 128kb vram. That is if the super nt has some free space left for the extra vram. My modchip was just a prototype but I could possibly do the final version this year.

paulb-nl, 2025-01-03:
It wasn't very clear but with software I meant software running on the SNES that uses 128KB VRAM.

Where can we find games that use it?

slidelljohn, 2025-01-03:
As far as I know no games use it. I have one game (mega man x 1) that I converted to use 128kb of vram but the game is still the same game. There is nothing new in the game but if you were able to view the vram you will see it’s set up for 128kb vram. I think bg 1 and 2 normally share 0x8000 bytes of vram but now they don’t share it. Bg1 uses its own 0x8000 bytes, bg2 uses its own 0x8000 and just those two already uses up the original 64kb vram. Bg3, sprites, and bg data use the last 64kb of vram(0x10000-0x1ffff).

Here is a link to one of my snes project pages that has a download in my first post that has this 128kb of vram modification for mega man x1. https://www.romhacking.net/forum/index.php?topic=24728.0

I was modifying a snes test cartridge to be able to use the 128kb of vram but I never finished it. I could do some more to it if there is interest in an updated test cart that supports 128kb vram. I did have it to where it could detect 64kb, 128kb, and 256kb(128kb x2 sets) of vram on real hardware using the hacked test cart. 

Anything else y’all need just let me know and I’ll get back to y’all as soon as I can. 

Also to know if your emulator or hardware is using 128kb of vram you should see some distorted graphics in yoshi’s island in the beginning of the game just before you start a quest. Some original games can still play normally with 128kb of vram as long as that games stores its vram in the 0x00000-0x0ffff area. If the original games stores vram anywhere in 0x10000-0x1ffff then odds are that game’s graphics could looked messed up like yoshi’s island. In 64kb system the extra bit is dropped and that’s why yoshi’s islands plays normal on original hardware. When 128kb is active the byte doesn’t get dropped causing yoshi’s island to try and use the full 128kb of vram  but it can’t because it wasn’t programmed for that but a patch can be made for it to work.

slidelljohn, 2025-01-03:
I wish those who maintain mesen 2 would add 128kb of vram support to the snes debugger. If we had that then others could start working on some projects that support the extra vram. I am considering adding 128kb of vram support to super mario world and super metroid if the ones that developed the editors for them would consider adding the support to their level editors if I make the 128kb patch for the games. I don’t hack or play ether of those so I’ll only do those 2 if someone else does the editors. Currently there is only 3 games that I am considering doing 128kb hacks and writing the entire level editors and those games are mega man x, gradius, and The 7th Saga.
I don’t have any interest in making level editors for any other game but there could be other games that I haven’t mentioned that I could add the 128kb vram to if someone programs the level editor for it.

slidelljohn, 2025-01-03:
Here is a byuu website archive where I decided to go ahead and try to make the 128kb modchip. 
https://web.archive.org/web/20190118203327/http://helmet.kafuka.org/byuubackup2/viewtopic.php@f=16&t=1559.html

I hope whoever reads this enjoys the discussion. I thought those pages were lost until I found them in the wayback machine. Unfortunately the last few pictures are not showing the last revision but it was a interesting discussion.