MiSTer-devel/SNES_MiSTer #392: Sufami Turbo: loading single-cart rom after a dual-cart rom

bikerspade, 2024-12-30:
When initially starting the SNES core, loading a concatenated rom with only Slot A (no Slot B) for **SD Ultra Battle - Seven Densetsu** is successful (the game correctly detects only slot A is populated).

Following this step, loading a rom containing both slots populated (**SD Ultra Battle - Seven Densetsu** + **SD Ultra Battle - Ultraman Densetsu**) also seems to work correctly.

However, following this step, if the original single slot A rom is loaded, the game sees both slots being populated which is unexpected. In order for the single slot rom to load correctly again, I have to do a soft reset of MiSTer from the OSD and then reload the SNES core.

srg320, 2024-12-30:
I don't quite get the nature of your problem. 
The initialization procedure for these games (Seven+Ultra/Ultra+Seven) requires running the game with each cartridge in turn. Both cartridges must be inserted into slot A for initialization. You can then insert both cartridges into their slots and play the game. 

I have made a menu option for this purpose (Hardware->Sufami cart swapping). Follow the steps below to initialize:
1. Start the core, do not run the game.
2. Set the "Sufami cart swapping" option to YES.
3. Start the game and wait for the message about slot B.
4. Set the "Sufami cart swapping" option to NO and reset the game via the core menu.
5. Make a backup save.

bikerspade, 2024-12-30:
The problem I am describing is assuming both slot B and slot A are initialized for SD Ultra Battle - Seven Densetsu + SD Ultra Battle - Ultraman Densetsu and that game is reset to load the rom where both slots are initialized. If this is then followed by loading a rom configured to only contain either Seven Densetsu or Ultraman Densetsu, the core seems to still see the other slot occupied, as though some previous state is still being retained.

srg320, 2024-12-30:
How do I reproduce the problem? How do I determine that the core still see the other slot? What do I need to look at?

bikerspade, 2024-12-31:
Here are the steps to reproduce the issue:
1) Start SNES core

2) Load an sfc rom that only contains "SD Ultra Battle - Seven Densetsu (Japan)" for slot A (slot B is absent).
[image]

3) Load an sfc rom that only contains "SD Ultra Battle - Ultraman Densetsu (Japan)" for slot A (slot B is absent).
[image]

4) Load an sfc rom that contains "SD Ultra Battle - Seven Densetsu (Japan)" for slot A, and "SD Ultra Battle - Ultraman Densetsu (Japan)" for slot B.
[image]

5) Load the sfc rom from step 2 that only contains "SD Ultra Battle - Seven Densetsu (Japan)" for slot A. You will see that the loading screen produced from step 5 matches that for step 4 and not step 2, indicating that the core is still seeing slot B populated.
[image]

6) Load the sfc rom from step 3 that only contains "SD Ultra Battle - Ultraman Densetsu (Japan)" You will see that slot B is still being populated from step 4 where only slot A is replaced, resulting in both slot A and slot B containing SD Ultra Battle - Ultraman Densetsu.
[image]

The single-ST roms are 2097152 bytes (2MB), containing the Sufami Turbo BIOS (expanded to 1MB) + ST game rom (expanded to 1MB).

The dual-ST rom is 3145728 bytes (3MB), containing the Sufami Turbo BIOS (expanded to 1MB) + slot A rom (expanded to 1MB) + slot B rom (expanded to 1MB).

bikerspade, 2025-01-01:
This seems to be fixed by https://github.com/MiSTer-devel/SNES_MiSTer/commit/cafbb80a186dfee3fdd2784d5b4c98368434c1fc

closed 2025-01-04 by bikerspade