MiSTer-devel/SNES_MiSTer #393: Sufami Turbo: black screen on swap with some SD Gundam Generation dual-cart combinations

bikerspade, 2025-01-01:
Steps to reproduce the issue:

1) Load SNES core (either latest stable 20241228 or latest unstable 20241231)
2) Set `Sufami Cart swaping` to Yes
3) Load _SD Gundam Generation - Axis Senki + SD Gundam Generation - Colony Kakutouki_
4) Slot A initializes and prompts message about Slot B being uninitialized (this is expected).
[image]

5) Open core menu and set `Sufami Cart swaping` to No. The game screen will immediately go black (not expected).
6) Reset the game via the core menu
7) A Sufami Turbo error screen will be displayed indicating a cartridge is inserted incorrectly.
[image]
[image]

The same issue will occur if you use the flipped version of this rom: _SD Gundam Generation - Colony Kakutouki + SD Gundam Generation - Axis Senki_

Combinations (and their flipped versions) that have the same issue:
_Axis Senki + Gryps Senki_
_Axis Senki + Ichinen Sensouki_
_Axis Senki + Zanscare Senki_
_Babylonia Kenkoku Senki + Colony Kakutouki_
_Babylonia Kenkoku Senki + Gryps Senki_
_Babylonia Kenkoku Senki + Ichinen Sensouki_
_Babylonia Kenkoku Senki + Zanscare Senki_
_Colony Kakutouki + Gryps Senki_
_Colony Kakutouki + Ichinen Sensouki_
_Gryps Senki + Zanscare Senki_
_Ichinen Sensouki + Zanscare Senki_

The others that are not mentioned seem to work OK.

srg320, 2025-01-03:
I have made changes in the operation of the Cart swapping option. Now when the option is enabled, cartridge B is placed in slot A, and slot B remains empty. This simulates the start of the console with cartridge B in slot A, as it is provided in the original. After initialization of cartridge B, the game starts as a single game. 
When game B has started, there is no need to return the option to the NO position. It is enough to do a RESET through the menu. The option will be reset, the game will start normally and cartridge A will be initialized.

srg320, 2025-01-03:
About the SD Gundam Generation games.
I don't know for sure how dual cartridges are supposed to work. But from what I found on the internet, not all cartridge combinations can work. There are three types of cartridges: blue, yellow and red. It seems that the yellow cartridge can't be the base cartridge. Correct me who knows.
[image]

bikerspade, 2025-01-03:
Interesting, is this base cartridge color rule documented in a wiki page somewhere?

srg320, 2025-01-03:
> Interesting, is this base cartridge color rule documented in a wiki page somewhere?

I just figured it out. And the game itself also points this out. But need confirmation from someone who is familiar with Sufami Turbo.

Also, with the latest changes, some combinations that didn't work before should work. For example, Colony Kakutouki + Axis Senki.

bikerspade, 2025-01-03:
> Also, with the latest changes, some combinations that didn't work before should work. For example, Colony Kakutouki + Axis Senki.

If so, that should now match the behavior of bsnes.

bikerspade, 2025-01-04:
Based on my testing, the following are the incompatible SD Gundam combinations, based on identical behavior between the [latest changes](https://github.com/MiSTer-devel/SNES_MiSTer/commit/5536103b652a927dead813e44b7cec9aac376267) in this core and bsnes:

Axis Senki + Colony Kakutouki
Axis Senki + Zanscare Senki

Babylonia Kenkoku Senki + Colony Kakutouki
Babylonia Kenkoku Senki + Zanscare Senki

Ichinen Sensouki + Axis Senki 
Ichinen Sensouki + Babylonia Kenkoku Senki 
Ichinen Sensouki + Colony Kakutouki
Ichinen Sensouki + Zanscare Senki

Gryps Senki + Axis Senki 
Gryps Senki + Babylonia Kenkoku Senki 
Gryps Senki + Colony Kakutouki
Gryps Senki + Zanscare Senki

With either Colony Kakutouki or Zanscare Senki in Slot A, any other cart can be in Slot B.
If either Gryps Senki or Ichinen Sensouki is present, Slot A should contain either one.

closed 2025-01-04 by bikerspade