MiSTer-devel/SNES_MiSTer #395: Unholy Night: The Darkness Hunter // Scrambled title screen

Zet-sensei, 2025-01-13:
Some users on Discord discovered that issue.

The title screen of that unlicensed game is scrambled on the last version of the core.

[image]

Max833, 2025-01-13:
This game has issues on hardware as well: https://github.com/SourMesen/Mesen-S/issues/144

srg320, 2025-10-30:
> I tested the title screen after pressing reset with an Everdrive X5 on 3 different SNES consoles. I did not see any glitches.

Thanks for checking.

paulb-nl, 2025-11-01:
It seems this issue is related to the Audio clock. 

https://github.com/MiSTer-devel/SNES_MiSTer/blob/a9c963e2a80567abd5fd534e5252e097db968019/rtl/DSP_PKG.vhd#L10

Changing the clock to 410496 fixes this issue. The DSP clock is then 32070.  Which is the average of bunch of consoles taken from here: https://github.com/MiSTer-devel/SNES_MiSTer/issues/208#issuecomment-593164533

srg320, 2025-11-01:
This explains the difference in timing when decoding the title screen graphics, which takes a long time. As a result, the IRQ handler (which is poorly designed) occurs during graphics decoding and corrupts the stack.

srg320, 2025-11-01:
I just measured the SCLK frequency on my console (at 50MHz). It is 4105740-4105815Hz.

srg320, 2025-11-01:
> I tested 2 of my consoles with the SMP clock speed measurement tool from here: https://forums.nesdev.org/viewtopic.php?t=24610
> 
> I got 32020 and 32046 at cold boot so I tried a core build with audio clock 409856 (32020) and that also fixes the issue.

I got 1026380 * 4 = 4105520Hz (32074Hz)

sorgelig, 2025-11-04:
There was a time when i tweaked the clock to value when sound of Bad Apple wasn't disappear. That was my test to find a correct CPU clock. May be it was a wrong way to fine tune, but i was thinking on real system sound shouldn't disappear.