MiSTer-devel/SNES_MiSTer #396: Enhance Auto/NTSC Video Region in DV1 to Improve HDMI Device Compatibility

jesusburnsneon, 2025-01-16:
Few people, including myself, have noticed the SNES core in [DV1 (direct_video=1)](https://github.com/MiSTer-devel/Main_MiSTer/issues/808) doesn't work with some HDMI switches connected to scaling devices like the RT4K and Morph 4K.

Well I recently discovered that SNES DV1 does work with my HDMI switch to RT4K and Morph 4K, but only if I set the core's Video Region to PAL.  If I flip it to Auto or NTSC, then it's a complete black screen and not even the Menu core appears.  Yet being restricted to PAL region limits rom options, especially any that force a region check.

I do not experience this region issue with any other console cores in DV1 (aside from certain roms with MegaDrive, which [may be related to this issue](https://github.com/MiSTer-devel/MegaDrive_MiSTer/issues/53)).  So I am wondering if the SNES core's timings/frequencies with the the Auto/NTSC Video Region in DV1 can be enhanced to improve compatibility with intermediate HDMI devices connecting to scalers.

birdybro, 2025-01-17:
Someone mentioned something similar for the NES core in discord, and I know way in the past another person that had an overly complicated switch configuration couldn't get it to work on any core.

When you remove the HDMI switch, does it work? If so, then maybe those switches are mangling the frequencies or messing with the DV1 packets at certain frequencies.

jesusburnsneon, 2025-01-17:
Hi birdybro!  Thank you for responding.  The Auto/NTSC regions do work without the HDMI switch (it's a TESmart 16x1), yet I figured I would ask in case there's a chance anything can be done with the SNES core considering when set to PAL region it works with the HDMI switch.  It's making me wonder about the Auto/NTSC region settings and if there's something with the timings/frequencies causing it to behave this way.  All of the other console cores I've tried with the HDMI switch show the menu core and run fine in any region.

Alternatively, and probably a broader discussion, I'm wondering if there's consideration to apply a setting in MiSTer.ini for the FX-Direct mode.  PixelFX came up with it many months back for xDigitals/GEMs going through HDMI switches/splitters/matrices to Morph 4K after similar issues were reported.  They published the [FXD InfoFrame metadata details](https://docs.pixelfx.co/VSIF-metadata-FX-direct.html) for anyone to use.  Including it as an option for MiSTer.ini would allow for a direct video setting that has better compatibility with HDMI devices.