MiSTer-devel/SNES_MiSTer #397: Crash: SMW Romhack - A Plumber For All Seasons

nol166, 2025-01-18:
## Purpose
While a large majority of issues with rom hacks were resolved with the most recent update, there is one that still seems to have issues. After starting the game, Mario's sprite lacks color and the game soon crashes a few seconds after starting a level. 

## To replicate
- Patch the [clean](https://github.com/videofindersTV/super-mario-world/) rom with the [patch](https://dl.smwcentral.net/28856/A_Plumber_For_All_Seasons_2021-11-22.zip)
- Run the game and start any level
- Try to get a powerup (fireflower, mushroom etc)

Tested with the following versions of the core:

```
SNES_20241228.rbf (current)
SNES_20240919.rbf
SNES_20240408.rbf
SNES_unstable_20241231_13a318.rbf
SNES_ss_20241125.rbf
SNES_ss_20241109
```

paulb-nl, 2025-01-19:
Works fine here with version 20241228 and 20240919. Try to rename your save file so it starts without a save. It could be corrupted.

nol166, 2025-01-20:
Ah okay, that might be the issue if it is working for you on `20241228`. I just deleted the `.sav` so I will report back.

nol166, 2025-01-20:
@paulb-nl it was in fact a corrupted save. Thank you for the info. I will remember this in future and add an update the issue in case anyone else runs into this.

closed 2025-01-20 by nol166