MiSTer-devel/SNES_MiSTer #399: [Possible regression] Recent commit breaks SMBC Demo (MSU-1)

vanfanel, 2025-02-17:
Hi there,

The recently released SMBCD demo (https://famicomcd.org/smbcd/download) is broken by this recent commit:
https://github.com/MiSTer-devel/SNES_MiSTer/commit/4502df9b828147e96f648d67409f2bbe0b989c51
After this exact commit, the game glitches out after choosing a level (ie: actual gameplay is impossible because it either glitches out and hangs, or simply resets the system after glitching out). 

Just before that exact commit, the game/demo runs fine.

srg320, 2025-02-20:
How to make a rom?

littlegamer87, 2025-02-22:
https://github.com/MiSTer-devel/SNES_MiSTer/commit/56f105c1f15c9267a25340e840bbf2703700e56e - please test and close if resolved

Zet-sensei, 2025-02-22:
Tested by multiple users & myself, should be closed.

vanfanel, 2025-02-22:
> Tested by multiple users & myself, should be closed.

Aye aye, Sir!

closed 2025-02-22 by vanfanel