MiSTer-devel/SNES_MiSTer #403: New release please?

littlegamer87, 2025-05-10:
Hi, looks like there is no activity on the unstable build for now (commits/bug reports). Current stable has SA1/MSU-1 issues, kindly request if a release can be considered to push these fixes to stable. Thank you!

littlegamer87, 2025-05-22:
That bug is already fixed in unstable a few months ago, and this is not about me. Myself and many can play SA1 games with the unstable just fine, and I see others asking support why they can't get -random SA1 game- to work. The latter is unnecessary if the fix is in unstable, very well tested 3 months since the commit. I think it's fair to speak up about these major bugs that should not exist in a stable version.

birdybro, 2025-05-22:
He will do a new release when he feels the time is right. People have the unstable release or the ability to compile their own if they so wish as an alternative in the meanwhile. There may be a valid reason for waiting that you are not aware of.

littlegamer87, 2025-05-25:
Indeed. Thank you for the consideration and support!

closed 2025-05-25 by littlegamer87