MiSTer-devel/SNES_MiSTer #404: Mega Man X3 crashes on title screen

Justturngreenpls, 2025-06-01:
On the latest stable build (2025-05-25) Mega Man X3 either freezes as its fading in to the title screen, or cuts to black shortly after fading in.  It works fine on the previous stable build (2024-12-28). Rom has the CRC32 hash of fa0fe671.

littlegamer87, 2025-06-02:
Can't reproduce. Both Megaman X2 and X3 (the only games that use CX4) run as expected on latest stable, CRC of the X3 rom is identical (USA). Tested on DE-10 HDMI with 1x128MB SDRAM.

Are you on DE10 or a clone?

Justturngreenpls, 2025-06-02:
I'm using an original DE10 from Terasic. Emudetect displays a screen saying "real hardware". I'll include a video demonstrating what happens when the games are loaded. There's a small edit at the end to show another way Star Fox 2 will crash with a blinking yellow screen. I tried Kirby Super Star and Super Mario RPG each for around 3 minutes and neither had any issues.
All roms play perfectly on the previous stable core.

https://github.com/user-attachments/assets/4ae50e4f-4007-4b3b-9121-7f524e0ba7b9

Justturngreenpls, 2025-06-02:
All previously tested games work in the unstable build from 2025-04-05. In the build from 2025-03-05 the CX4 games both crash on a black screen after the Capcom logo, the SA1 and GSU games seem to work fine. The 2025-02-22 build boots the GSU and CX4 games to a black screen, but plays the SA1 games fine.

littlegamer87, 2025-06-02:
Those results from 2025-02-02 are explainable, the rest are not. The only difference between stable and last unstable commit is a sys update which cannot explain a CX4 issue, and neither can version 2025-03-05. I don't think the answer to your issue is in this repository.

Unless there is another explanation (eg bad binaries, which I can't confirm either) you might be looking at early signs of hardware failure.

TheJesusFish, 2025-06-03:
Ok, I've done some testing on this and can confirm that it is an issue with _some_ MiSTers.

**Mister 1** (de10 nano, dual ram, direct video to morph4k):
- MegaMan X2 boots and gets into gameplay fine
- MegaMan X3 boots and gets into gameplay fine
- Starfox 2 boots and gets to the main menu but hangs there (music keeps playing)

**Mister 2** (Mister Pi, dual ram, direct video to tink4k)
- All games load and play fine

**Mister 3** (Mister Pi, dual ram, HDMI to iPad screen)
- All games load and play fine

**Mister 4** (de10 nano, dual ram, HDMI to iPad screen)
- Megaman X2 will get to various states of boot (sometimes just the cap com logo, sometimes the title, sometimes it plays all the way through)
- Megaman X3 doesn't boot (not even the cap com logo)
- Starfox 2 boots sometimes but always yellow (or blue) screens before the menu

**Mister 5** (de10 nano, dual ram, direct video to CRT)
- Megaman X2 is fine
- Megaman X3 is fine
- Starfox 2 crashes occasionally - cannot figure out the trigger

**Mister 6** (Qmtech, single ram (soldered on), HDMI to iPad Screen)
- Almost the exact same results as Mister 4

I built a fresh core from GitHub, and none of the games boot on that core (on any system). Other games boot fine on that core.

The unstable from April 4th fixes the issue on every mister EXCEPT Mister 6 (QMtech) which still won't let me play star fox without yellow screening.

So yeah, feels like a transient issue of some sort. Reminds me of the weirdness the Jaguar core had right after grey rewrote it (before he locked down stability).

Happy to help test more.

littlegamer87, 2025-06-05:
I hate to bring it to you but finding odd behavior on an FPGA when it is running hot is almost certainly hardware damage.

littlegamer87, 2025-06-05:
Just tried it out at your request, and after exceeding 90 minutes of Doom 2 demos on 486 profile I switched to Starfox 2 and defeated those incoming missiles without crashes or issues, as expected. I still stand by what I said that an fpga that is malfunctioning when hot is (almost) certainly hardware damage, and if your failures are that early, I also believe that this is likely esd damage and that you repeated the mistake on all of them. Sorry, but I still think this repository is not the right place for answers to these issues.

I do get the idea that Starfox 2 can serve as a good mister self test. There is probably not a lot of unused die space left when that game is running, and the cycle accuracy makes malfunctions more noticeable. Bringing heat into the test was a good idea. Sorry about the result.

TheJesusFish, 2025-06-06:
@Justturngreenpls When you have a second, try this build. I haven't been able to reproduce the issue with it (it's the current core built from seed 10).

[SNES10.zip](https://github.com/user-attachments/files/20619837/SNES10.zip)

Justturngreenpls, 2025-06-06:
@TheJesusFish All previously nonworking games are perfectly functional on the core provided.

Zet-sensei, 2025-06-09:
I can confirm the same behavior on my MiSTer systems with official DE10 Nano (Single and Dual).

The last save state core from @paulb-nl branch refuses to work on Mega Man X2/Rockman X2 -with a standard seed- on mine (other games cited seem fine).
Same for the last stable core here, like the others explained already.

No problem on the QMTECH & MiSTer Pi clones, with the same game files & cores during my tests today.

If I try to compile the stable core once more with another seed, the game works once again.
I can't compile the last save state core so far as paulb-nl didn't updated the branch publicly with the last changes (like with The Firemen fix).

Maybe he could see that message here and recompile/share the core with a seed 10 (for example) and/or update the code the the 'save state' branch later.

TheJesusFish, 2025-06-14:
@Justturngreenpls the build should be live now in updaters. you can close this once you confirm.

TheJesusFish, 2025-06-14:
@Justturngreenpls try this build when you get a chance

[SNES11.zip](https://github.com/user-attachments/files/20738229/SNES11.zip)

wwark, 2025-06-14:
Hi @srg320 for information @kandowontu confirmed to me the option  hardware -> SuperFX FastRom is not used anymore on Star Fox EX rom hack. Normally it is not used to other rom hack.
I don't know if it could help this issue but it could help to save few logics.
Thanks,

Justturngreenpls, 2025-06-14:
@TheJesusFish The post is gone now but I tried your seed 11 core. I tested it with all previously aforementioned games and the remainder of the commercially released SuperFX games, and just for thoroughness Star Fox EX. All worked perfectly in that new core, I completed a level in each to ensure stability.

@littlegamer87 If this was a hardware malfunction it would effect more than just two enhancement chips in one specific core. And it wouldn't happen the same way across multiple devices.

TheJesusFish, 2025-06-14:
Yeah was my bad. Should have a new build here in like 10 min.

Justturngreenpls, 2025-06-14:
@TheJesusFish In the previous SNES11 build I was able to complete a level in Yoshi's Island without issue, with this one though the game freezes just after loading into the level. Which is slightly different behavior to the SNES10 build where it freezes just before the screen transition effect. Picture for where exactly it froze.

[image]

TheJesusFish, 2025-06-14:
@paulb-nl Yeah, the last unstable actually works fine (non-save state - I havent tested the save state core much)

@Justturngreenpls try this pack and see if any of these builds fix it 

[SNESPack.zip](https://github.com/user-attachments/files/20741999/SNESPack.zip)

TheJesusFish, 2025-06-15:
gonna have the discord test seed 13. all the games I tested on it looked good. Thanks for testing @Justturngreenpls

littlegamer87, 2025-06-15:
> [@littlegamer87](https://github.com/littlegamer87) If this was a hardware malfunction it would effect more than just two enhancement chips in one specific core. And it wouldn't happen the same way across multiple devices.

As I tried explaining, results will be random. On a good fpga, whichever seed you choose should have no effect, provided timing constraints are met and there is no core issue. You _will_ find issues with some other (large) cores if you keep testing enough, but this does not have to be crashes. It can also be graphics (layering/glitches) or audio pops/distortion. The SNES seems more prone to crashing when things are out of place.

You did not specify which games are crashing on core 8, but below are some results of mine (Yoshi's Island, Star Fox 2, Megaman X3). 
https://we.tl/t-Kd19DJk6Pw - expires in 3 days

littlegamer87, 2025-06-15:
Fine, believe what you want to believe. This type of damage will get worse over time and typically leads to a device unusable or a brick in a time span of ~3-4 years. When that time comes, I hope you will reflect on this conversation.

Justturngreenpls, 2025-06-15:
Yeah yeah, enough of your unhelpful FUD.

paulb-nl, 2025-06-15:
@Justturngreenpls are you able to test with a different SDRAM? I am curious if that makes a difference.

I am trying builds with different seeds to find one that crashes on my De nano so I can debug it. Unfortunately all builds work fine (lol).

Justturngreenpls, 2025-06-15:
@paulb-nl Unfortunately I only have one SDRAM module. It's revision 2.9

TheJesusFish, 2025-06-15:
@paulb-nl Not sure if I can be helpful in some way but I have 4 misters that I can repro this on.

- de10nano dual ram misteraddons stack in the passively cooled aluminum armor
- de10nano dual ram misteraddons stack with no case (printed out [this top and bottom](https://www.printables.com/model/79238-mister-fpga-top-and-bottom-plate-case), and am running a fan on the io board) 
- de10nano dual ram retrocastle digital stack with a fan
- QMTech stack passively cooled with a heatsink

My MisterPis, funnily enough, are totally fine.

@Justturngreenpls what do you have plugged into your stack? How are you powering it?

paulb-nl, 2025-06-15:
@TheJesusFish Try to swap the single ram stick between systems that have different/no issues and see if the results change.

paulb-nl, 2025-06-15:
Thanks for testing this. Sadly still no definite answer.

Would it normally crash with graphic corruption like that or is it new behavior?

This happened with the game left idle right?

paulb-nl, 2025-06-15:
One thing that CX4 and Super FX have in common is that they have the same exact faster ROM read signal.

I remember I had same issues with the new SDRAM controller while working on the save state core which was related to reads and writes being too close together.

I added a branch where I put the WRAM back into BRAM to avoid writes into SDRAM.
https://github.com/paulb-nl/SNES_MiSTer/tree/test_wram

Can you compile some builds from that branch with different seeds and try if it works better this way?

TheJesusFish, 2025-06-15:
I'll start right now :)

TheJesusFish, 2025-06-15:
@paulb-nl 

Despite not meeting timings, Seed 1 worked with everything I thew at it. Trying to find a build that meets timings.

If you want to try @Justturngreenpls 

[SNES_Seed1_BadTimings.zip](https://github.com/user-attachments/files/20747520/SNES_Seed1_BadTimings.zip)

Justturngreenpls, 2025-06-15:
@TheJesusFish Both CX4 games, Star Fox 1&2, Stunt Race FX and Yoshi's Island all appear to be working on that core.

paulb-nl, 2025-06-16:
The builds that I was trying before and were working fine were seed 1-8 and they all failed timings.

If the builds with WRAM in BRAM are working fine for you with different seeds then we found the issue.

TheJesusFish, 2025-06-16:
@Justturngreenpls You already tested seed 1. Here is 2-5. I'll test 3 and 4 in the morning.

[SNES_Wram-2-5.zip](https://github.com/user-attachments/files/20750630/SNES_Wram-2-5.zip)

Justturngreenpls, 2025-06-16:
@TheJesusFish All of these seeds work for SuperFX and CX4 except seed 3, which freezes in SuperFX games but works with CX4.

TheJesusFish, 2025-06-16:
@paulb-nl Did a full test run of seeds 1-5 on two of my problematic misters (a de10 and the qmtech). No crashes or hangs at any point.

Testing run was:
- SMW2: play through the tutorial and then start the first real level
- Star Fox 2: start an interception mission
- Mega Man X3: play through the first ladder transition
- Mega Man X2: destroy the first couple of enemies

I don't know anything about anything, but is this issue similar to what robert faced with the n64 core and batch 1 mister pi ram? Basically, the core was reading the sdram before it was ready, so he had to push the read cycle a little bit to make it compatible. If that sounds helpful, I can pull up the discord stuff about it - I just wasnt sure if you wanted to go all in on the bram route :).

@Justturngreenpls I wonder if pulling re-seating your memory will fix seed 3. It could be that there is also something there playing a part, and that would rule that out.

srg320, 2025-06-16:
Did you have any problems with SDD1 games on "bad" builds?

TheJesusFish, 2025-06-16:
I don't think so.

I just did a full set of SFA2 on the current stable core

Star Ocean seems to be working too (but I am still just watching the intro)

TheJesusFish, 2025-06-16:
Could another option just be to use Seed 13 going forward? I believe that worked for @Justturngreenpls and me.

Toryalai1, 2025-06-16:
I agree with Paulb-nl. Changing seed is not the way forward. We need to find the root cause.

TheJesusFish, 2025-06-16:
I know nothing. I just have an irresponsible amount of misters and the inability to make good financial decisions.

If there is more testing to be done, let me know :)

littlegamer87, 2025-06-18:
> Of course I say that and then turn around to this:
Are your other setups similar to this, with no enclosure and a crt for extra charge? In that setup, you will zap your mister very frequently. 

I do not recommend investigating tests performed on this. If the heat remark wasn´t convincing enough, then surely this picture is.

@paulb-nl I don´t think this is an sdram issue. By logic, an issue there would also needed to be present in last stable, but that build was confirmed by both as fine, think sdram behavior is the same. I think this is triggered by the extra logic committed incl sys that is now making some fpga malfunction very noticeable. Different seeds 'work' so allocation matters. CX4 and GSU also have multipliers in common. 

Why are you calling the core 'very unstable´? Can you reproduce?

birdybro, 2025-06-18:
If different seeds produce different results that usually points to a marginal timing issue that the timing analyzer can't even pick up. This typically can start happening around 80% or more logic utilization, but other things can cause it. That's what paulb-nl means by unstable. The seed shouldn't impact the behavior of the logic. Also, yes, changing seeds is a temporary workaround that will have different results for different people who have a different bit of luck in the silicon lottery.

littlegamer87, 2025-06-19:
@birdybro I find it hard to make a case for a 'silicon lottery' or 'luck' if this user has just shown a complete disregard for static. It isn't luck if you walk away unharmed if you decide to not jump off a bridge.

TheJesusFish, 2025-06-24:
@paulb-nl Interesting. I wonder if it's the placements of the build because everything (Yoshi, SFA2, SF2, MMX2, and MMX3) boot and play fine here.

littlegamer87, 2025-06-25:
@TheJesusFish probably. Try some different seeds again to know for sure

Justturngreenpls, 2025-06-25:
All CX4 and SuperFX games are working on the latest unstable (2025-06-24) for me.

TheJesusFish, 2025-07-10:
Could DE10 board revision be a factor here? Here is what I can find so far about the different revisions of the board:

[image]

All 3 of my DE10 nanos (which are affected by this bug) were bought in the last year. Folks on the discord who aren’t having trouble are saying that they have had their DE10 nanos for years - so could a hardware revision be a factor?

Unfortunately, I haven’t yet found a way to determine hardware revision as the spec sheet I got that information from only listed how to identify hardware revision A, B, and B2. I intend to dive into that more today.

Another interesting hardware quirk is that the MiSTer Pi based board may be unaffected by the bug. We have tested around a dozen MiSTer Pi boards, and none of them have trouble. That isn’t a statistically significant number yet, but it’s definitely odd as I have 4 MiSter Pi’s and none of them are affected. Taki is going to use CX4 games as test roms in the batch 4 mister pi testing, so we should have statistically significant data soon.

paulb-nl, 2025-09-01:
Me and @TheJesusFish have been debugging Megaman X2 freezes on his hardware.

We narrowed it down to SDRAM sometimes returning data from the wrong address.

For example, here is a SignalTap trace where the core requests data from SDRAM address $5F16 and the SDRAM returns data from address $5D16. The value should be $C8DF but it returns value $0001. 

[image]

What is interesting is that all these wrong reads happen when the CPU is executing JSR ($ADDR, X) instruction and reading the $ADDR+X data. The above wrong read happened during the instruction below. It then jumps to the wrong address and usually causes a BRK interrupt and goes in to an infinite loop.

[image]

I made a patch for the X2 rom that shows a stack dump on screen when it goes to the BRK interrupt handler. With the stack dump it can be usually traced to an JSR ($ADDR, X) instruction gone wrong.
Patch: [mmx2_crash3.ips.zip](https://github.com/user-attachments/files/22083016/mmx2_crash3.ips.zip)

I only got one build of the core that had the same issue on my DE10 nano. This issue did not happen with normal cartridge games.

This issue may be related to the SDRAM clock output. Maybe Quartus does not know that the signals to the SDRAM need to be stable after half a clock cycle instead of a full cycle because those are not specified in the SDC file. Although is it very weird that it only happens to CX4 and other special chip games.

srg320, 2025-09-02:
> For example, here is a SignalTap trace where the core requests data from SDRAM address $5F16 and the SDRAM returns data from address $5D16. The value should be $C8DF but it returns value $0001.

Have I understood correctly? The addresses are different in different cases, but bit 9 of the address (addr0[9]) is always incorrect. This is obtained when comparing the read data and the ROM data.

> This issue may be related to the SDRAM clock output. Maybe Quartus does not know that the signals to the SDRAM need to be stable after half a clock cycle instead of a full cycle because those are not specified in the SDC file.

This would be correct if SDRAM_A were a wire, not a register.
You need to look at SDRAM_A[8] in Map Viewer/Chip Planner (with SignalTap turned off) and make sure that it is indeed a IO port register.

littlegamer87, 2025-09-02:
> I only got one build of the core that had the same issue on my DE10 nano.
Hi Paul, can you share this build? I think I should check if it fails on my end too.

littlegamer87, 2025-09-02:
Thank you Paul. Finished the prologue and one stage up until the boss, it did not crash.

sorgelig, 2025-09-04:
SDRAM signals MUST be registers in order to follow timings. Correctly designed SDRAM controller will have inferred register on output signals which will be placed directly to IOBUF by fitter and it will have minimal slack time. Also it won't have variation from build to build.

paulb-nl, 2025-09-04:
Your stack dump shows it crashed after the same instruction as in my Mesen screenshot above. The BRK interrupt return address is 00E628 so the BRK instruction was at 00E626. 

It looks like it read $DF15 correctly ($23) but it read $DF16 from $DC16 ($E6) so it jumped to address $00E623 instead of $00DF23. 

If you want to recreate this then you can use my .qsf and .stp file to build with SignalTap with the settings I used. The build I posted was with SignalTap on the SDRAM clock so it had big timing violations.

[SNES_stp.zip](https://github.com/user-attachments/files/22138090/SNES_stp.zip)

It could be that SignalTap shows some registers with different value than what the SDRAM gets because it duplicates registers for SignalTap I think.

littlegamer87, 2025-09-05:
FWIW I still believe the root cause is FPGA malfunction due to ESD damage. I can't think of another rationale why my build and all Mister Pi's are not affected - that information is actually very helpful. I now believe that the Mister Pi is the only prebuilt clone worth recommending, Taki seems to know what he is doing.

I suspect ao486's seed dependency has the same root cause, although that architecture does not lead to crashes but audio/video glitches. I haven't gotten my hands on enough 'bad' ao486 builds to say with confidence tho