MiSTer-devel/SNES_MiSTer #421: Potential Async Reset Deassertion Metastability Issue

rianhunter, 2025-08-11:
Hi @sorgelig 

I'm reading through the core because I plan to add a feature and while implementing my feature I noticed something in this core. Most of the code assumes an asynchronous reset. That's fine but the reset signal should be be synchronous when it deasserts since the underlying logic cells are only support asynchronous reset assertion (not deassertion). More can be read here: https://www.intel.com/content/www/us/en/docs/programmable/683082/25-1/using-asynchronous-resets.html

The usual fix is to make the reset deassertion synchronous (but keep the assertion asynchronous) is this:

```Verilog
reg [1:0] sync_ff;

always @(posedge clk or negedge async_reset_n) begin
    if (!async_reset_n) begin
        sync_ff <= 2'b00;  // Assert sync reset immediately on async reset assert
    end else begin
        sync_ff <= {sync_ff[0], 1'b1}; // Shift in 1's on each clock cycle after reset release
    end
end

assign sync_reset_n = sync_ff[1];
```

There is no code like that for the reset signal, so there is risk of metastability of the entire system when reset is deasserted.

There is some code to make the reset synchronous for the sdram:

```SystemVerilog
reg RESET_N = 0;
reg RFSH = 0;
always @(posedge clk_sys) begin
	reg [1:0] div;

	div <= div + 1'd1;
	RFSH <= !div;

	if (div == 2) RESET_N <= ~reset;
end
```

This code, however, is not safe. Since `reset` can change at any point, `RESET_N` risks becoming metastable. The right way to fix this would be to use the technique above on the reset signal e.g.:

```SystemVerilog
reg [1:0] sync_reset;
always @(posedge clk_sys) begin
    sync_reset <= [sync_reset[0], reset];
end
```

And then replace `reset` in the code with `sync_reset` everywhere. What do you think about that? Do you agree? Would you accept that PR?

sorgelig, 2025-08-11:
Are you talking about RESET_N get split into different registers after routing and may have delay in some paths? In most cases reset signal is not critical, so current way should work fine already. 
May be @srg320 have additional thoughts about how this change would be useful for SNES core.

birdybro, 2025-08-11:
Regarding your code snippet:

```sv
reg [1:0] sync_reset;
always @(posedge clk_sys) begin
    sync_reset <= [sync_reset[0], reset];
end
```

You used the wrong syntax for concatenation here, it should be `{}` instead of `[]`.

sorgelig, 2025-08-12:
> Not talking about that problem specifically. It's the problem in the [URL I linked](https://www.intel.com/content/www/us/en/docs/programmable/683082/25-1/using-asynchronous-resets.html).

Then it's not applicable to this core. SNES uses synchronous reset.

rianhunter, 2025-08-12:
> Then it's not applicable to this core. SNES uses synchronous reset.

Do you mean the original hardware resets synchronously?

There are various parts of the core that assume an asynchronous reset, e.g.:

https://github.com/MiSTer-devel/SNES_MiSTer/blob/13bf006c239e3dab044ac2fb6c6041602037a8b6/rtl/CPU.vhd#L425-L441

And the `RESET` signal from the top-level emu module is asynchronous:

https://github.com/MiSTer-devel/SNES_MiSTer/blob/13bf006c239e3dab044ac2fb6c6041602037a8b6/SNES.sv#L26-L28

Second level `reset` signal is derived from `RESET`, it's also asynchronous:

https://github.com/MiSTer-devel/SNES_MiSTer/blob/13bf006c239e3dab044ac2fb6c6041602037a8b6/SNES.sv#L288

And `RESET_N` that is fed into the main module is a synchronized version of `reset`:

https://github.com/MiSTer-devel/SNES_MiSTer/blob/13bf006c239e3dab044ac2fb6c6041602037a8b6/SNES.sv#L743-L752

The problem here is that the `reset` signal used to produce `RESET_N` isn't synchronized correctly because it isn't guaranteed to be stable before the positive edge of `clk_sys`. You need to do the trick listed in the URL, which is to buffer the asynchronous reset by at least two "follower" registers. In practice, metastability is rare because this signal only changes polarity once when the core runs but it still seems to be an issue.

---

My referencing the URL itself might be partly confusing since ultimately the reset signal is attempted to be synchronized in this core. If the reset signal wasn't synchronized, that would be half OK since the core seems to be designed for asynchronous reset assertion but the deassertion would still need to be synchronized and that's where the URL document comes into play.

birdybro, 2025-08-12:
What bugs or problems have you observed that go away when you make your changes and test the core?

rianhunter, 2025-08-12:
> What bugs or problems have you observed that go away when you make your changes and test the core?

I'm developing a new feature for the core and when I looked into what reset discipline I should follow in my modules ( asynchronous vs synchronous) I noticed this issue.

Theoretically, the only issue that can arise from this is metastability whenever reset is initially deasserted and the core starts. So it's not a practical issue in most cases but it's still a correctness issue.

Another issue is the consistency and discipline in terms of how the reset signal is handled in the codebase. This matters for new modules or features that are added. It would be ideal if there were an unambiguous reset discipline being followed so that new modules could implement it and it would be more clear whether or not the reset signals were behavior according to that discipline. This is more of an abstract issue but this is the issue that affected me.

I raised this issue because I did want clarification on what assumptions were being made about how the reset signal is expected to behave. I see evidence of different assumptions being made in different parts of the codebase, so I wasn't sure what the original intent was. Originally it seems that reset was expected to be asynchronous but then it seems things were changed to expect it to be synchronous, although I'm still not 100% sure about that because the change made doesn't look correct but maybe it is for reasons of which I'm not aware.

rianhunter, 2025-08-14:
> Probably, i don't remember already. May be it wasn't me :)

It's git blamed to you :) No problem, I'll experiment with that and add a comment when I figure out why it's necessary. I'll close issue for now but I will follow up with a PR in the future.